robotology/mesh-support #1193: Jerky motion on the right ankle pitch joint of iCubGenova09 (iCub3)

S-Dafarra, 2021-09-20:
### Device name 🤖

iCubGenova09

### Request/Failure description

The right ankle pitch joint does not seem to move smoothly. I suppose the motor encoder is dirty.

### Detailed context

After mounting the feet covers, with @Mick3Lozzo and @sgiraz we noticed that the right ankle pitch does not move smoothly. I think it could be the motor encoder, also considering that that joint still has the old cap for closing the motor hole

[image]

https://user-images.githubusercontent.com/18591940/134051979-822f5f87-4cc8-408f-9460-81f4f24b8c47.mp4

https://user-images.githubusercontent.com/18591940/134051986-8cbdbc5b-bbad-4624-bb1c-af4c692d3f4a.mp4

### Additional context

In any case, we recently used the robot and we did not notice any major issue.

cc @pattacini @GiulioRomualdi @DanielePucci

pattacini, 2021-09-20:
We had some trouble with the feet covers and @Mick3Lozzo and @sgiraz had to deal with that this afternoon; thus, we didn't finish our tests. That's why we booked the robot for tomorrow morning.

We can then do our checks.
Thanks for reporting.

cc @davidetome @sgiraz

Mick3Lozzo, 2021-09-21:
We cleaned the optical disk of the right ankle pitch joint, and this was the first result: the motor resulted completely blocked.

https://user-images.githubusercontent.com/15085666/134139080-45f0b75e-f791-406f-93d9-a793ad2ae671.mp4

We re-cleaned the optical disk, and this time the ankle pitch moved and did the jerky motion again, but in a different angular position respect to the previously observed one. 

cc @S-Dafarra @pattacini @davidetome

maggia80, 2021-09-21:
I think it's still a problem with the cleanness of the disk.

davidetome, 2021-09-21:
> I think it's still a problem with the cleanness of the disk.

@maggia80 we cleaned it few times, may we need to replace the sensor/optical disk as already done in the past?

maggia80, 2021-09-21:
I'll try one more time by using a lot of liquid. It will take a while to evaporate but it could solve. Next step will be to change the disk yes.

pattacini, 2021-09-21:
Since replacing the optical disc might be quite unwieldy in terms of downtime, and given that the joint oscillated only once, we decided together to proceed without the change in order to carry out at least a few experiments.

Keep the issue in radar 📡