robotology/mesh-support #1195: CRC error detected on the torso at startup on iCubGenova09 (iCub3)

S-Dafarra, 2021-09-20:
### Device name 🤖

iCubGenova09

### Request/Failure description

I have noticed that at startup, there are some CRC errors coming from the torso

### Detailed context

[log_icub-head_yarprobotinterface_1432_CRC_encoderReaderError.txt](https://github.com/robotology/icub-tech-support/files/7197783/log_icub-head_yarprobotinterface_1432_CRC_encoderReaderError.txt)

pattacini, 2021-09-20:
See https://github.com/robotology/icub-tech-support/issues/1193#issuecomment-923229854.

cc @davidetome @sgiraz @marcoaccame

davidetome, 2021-09-21:
As per the right leg, I ran few times torso and no errors came up

@marcoaccame

marcoaccame, 2021-09-21:
i had a look at the log. 
yes, there are two crc errors detected both by board 10.0.1.5. they are about 1 second apart. 
each one has caused the loss of a single packet coming from board 10.0.1.7 and directed to icub-head
here is diagnostics about one of the lost packets.

``` 
hostTransceiver()::parse() detected an ERROR in sequence number from IP = 10.0.1.7. 
Expected: 3966, Received: 3967, Missing: 1, Prev Frame TX at 45983180 us, This Frame TX at 45993181 us 
 
from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 46s 9m 636u: (code 0x06000002, par16 0x0001 par64 0x3000000000000001) -> ETH monitor: detected RX CRC error in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal).  Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. Number of errors in par64&0xffffffff + . 
```

And yes, there are two messages about AEA spikes coming from the same board. at different times, however. so they are surely not related to a unique episode.

pattacini, 2021-11-23:
@S-Dafarra still reports CRC errors on the torso at startup.
We may need to investigate it further. Most likely, after all the upcoming deadlines.

Fabrizio69, 2022-01-18:
Hi @pattacini , iCub3 is still under maintenance with various solved problems and other under test. 
I think that it's better another test for these glitches (we changed the wiring).

cc: @emilianob80 @Uboldi80

Uboldi80, 2022-01-18:
Now the problem with the torso is the sequel:
[image]

emilianob80, 2022-01-19:
This morning after a check I found these two potential causes of the problem:

**_Sensor WS0 -- Red wire on male connector side badly crimped_**

[image]

_**Broken CAN termination resistor**_

[image]

I have proceeded to fix the problems. We just need to test the operation @Uboldi80 

cc: @pattacini @S-Dafarra @Fabrizio69

emilianob80, 2022-01-19:
After running a functional test with @ Uboldi80 the problems are solved. The issue can be closed

cc: @pattacini @S-Dafarra @Fabrizio69

pattacini, 2022-01-19:
Hi @emilianob80 

Thanks heaps for sorting out the problem that was preventing the torso from starting up.

Actually, the issue was originally meant to address those CRC errors spotted at the robot launch by @S-Dafarra. Therefore, if you don't mind, I would keep it open 'till @S-Dafarra will confirm they're gone.

emilianob80, 2022-01-19:
> Hi @emilianob80
> 
> Thanks heaps for sorting out the problem that was preventing the torso from starting up 👍🏻
> 
> Actually, this issue was originally meant to address those CRC errors spotted at the robot launch by @S-Dafarra. Therefore, if you don't mind, I would keep it open 'till @S-Dafarra will confirm they're gone during their operations.

Absolutely no problem 👍🏻

S-Dafarra, 2022-03-15:
Now the CRC happen also after startup. See https://github.com/robotology/icub-tech-support/issues/1321 and https://github.com/robotology/icub-tech-support/issues/1327

S-Dafarra, 2022-04-11:
Interestingly, during the tests at Ministero della Salute, we noticed that the CRC occurs while walking, but not when moving the joints manually or when touching the cables. Moreover, @gsisinna noticed that one of the 2FOC connected to the torso ems gets suspiciously hot during walking. I don't know if this could be related

pattacini, 2022-04-12:
> but not when moving the joints manually 

Does this mean via the GUI?

S-Dafarra, 2022-04-12:
> > but not when moving the joints manually
> 
> Does this mean via the GUI?

Yes. We also tried to move all the leg joints in idle, but it was not happening.

gsisinna, 2022-07-28:
@S-Dafarra after the latest robotology-superbuild release and the new icub-firmware-build version are you still observing these problems?

S-Dafarra, 2022-07-28:
> @S-Dafarra after the latest robotology-superbuild release and the new icub-firmware-build version are you still observing these problems?

Hi @gsisinna, actually we haven't checked to be honest. Thanks for the reminder

S-Dafarra, 2022-09-06:
We recently tested the walking too and the CRC on the torso seem gone. Only a few of them have been detected on the left leg

[log_icub-head_yarprobotinterface_8183.txt](https://github.com/robotology/icub-tech-support/files/9497986/log_icub-head_yarprobotinterface_8183.txt)

pattacini, 2022-09-06:
Great!

Perhaps, @marcoaccame can have a look a those sporadic CRC error on the left leg (I've just counted 3 of them).

marcoaccame, 2022-09-08:
Yes, there are three situations in which the board `left_leg-eb8-j3_5` detects one single CRC error in its output port. But I have also seen that `yarprobotinterface` sometimes detects missing frames from `10.0.1.1` and `10.0.1.2` (progressive number is not consecutive). I have also seen that in one situation the CRC detection and the missing frame are close in time. 

I will better verify time proximity of these two phenomena, but the only action I could recommend does not change:

check the cables in connector `P3` of the  board `left_leg-eb8-j3_5`.

pattacini, 2022-09-08:
>The only action I could recommend does not change: check the cables in connector P3 of the board left_leg-eb8-j3_5 and maybe recrimp.

Thanks @marcoaccame 👍🏻 
Perhaps, @emilianob80 this is something that we can quickly do during the next maintenance.

closed 2022-11-06 by pattacini