robotology/mesh-support #1196: ``CFG: EOtheEncodeerReader cannot be configured: verification fails.`` error at startup on iCubGenova09 (iCub3)

S-Dafarra, 2021-09-20:
### Device name 🤖

iCubGenova09

### Request/Failure description

When starting the robot we receive this error
```
from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src LOCAL, adr 0, time 26s 734m 549u: (code 0x05000016, par16 0x3001 par64 0x000000000fff0009) -> CFG: EOtheEncodeerReader cannot be configured: verification fails. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + . 
```
Not sure what this means @pattacini @marcoaccame 

### Detailed context

[log_icub-head_yarprobotinterface_1432_CRC_encoderReaderError.txt](https://github.com/robotology/icub-tech-support/files/7197792/log_icub-head_yarprobotinterface_1432_CRC_encoderReaderError.txt)

pattacini, 2021-09-20:
See https://github.com/robotology/icub-tech-support/issues/1193#issuecomment-923229854.

cc @davidetome @marcoaccame @sgiraz

davidetome, 2021-09-21:
I ran few times `yarprobotinterface` on the right leg but no errors came up.
@marcoaccame any thoughts?

```

marcoaccame, 2021-09-21:
> Not sure what this means @pattacini @marcoaccame

it means that the primary encoder on joint #0 of board 10.0.1.11 ((right_leg-eb11-j0_2)), which is an AMO, has failed its reading at startup.
for this reason the motor control service cannot should not started.

> I ran few times yarprobotinterface on the right leg but no errors came up.
@marcoaccame any thoughts?

I think that sometimes the errors on the AMO encoder happen at the startup of the service (as in this case) where the design of the system verifies the availability of every required resources and if any resource is missing or not working correctly yarprobotinteface is informed.

In some other cases errors on AMO don't happen or happen later.

pattacini, 2021-09-21:
We briefly discussed these observations and said that they seem isolated events.
Will be monitoring the evolution.

pattacini, 2021-11-23:
Never seen again.
Closing as agreed.

closed 2021-11-23 by pattacini