robotology/mesh-support #1198: Right thumb oppose  tendon broken - iCubGenova09

davidetome, 2021-09-21:
### Device name 🤖

iCubGenova09

### Request/Failure description

the right thumb oppose is loose, probably due to a broken tendon

cc @pattacini 

### Detailed context

/

davidetome, 2021-09-23:
@GiacomoVassallo13 is going to fix this 🛠️

GiacomoVassallo13, 2021-09-23:
@davidetome the thump oppose was been replaced.
Ticket:
cc: @Fabrizio69 @Uboldi80

isorrentino, 2021-09-28:
I just turned on the robot and the thumb of the right hand moves without any input and continuously. Reopening.
cc @pattacini

pattacini, 2021-09-28:
cc @GiacomoVassallo13 @davidetome

pattacini, 2021-09-29:
Moved the writeup of @Uboldi80 from https://github.com/robotology/icub-tech-support/issues/1058 here, as this is the originating issue:

> for some reason that I still do not fully understand, despite the pulling of the hand tendons being performed at least 2 more times, the tendon retaining grains were again slack. this obviously contributes to making the joint difficult to control as well as making it oscillate. This morning I re-tensioned the thumb cables but this time I added a special low-resistance bellow to prevent damage by irreparably blocking the grain. it is loctite 222 pink. probably part of the problem is due to the size of the fixing screws: I believe that M 1.6 is really too little to offer adequate resistance, also considering that the threaded piece I believe is in aluminum, so I assume that following some uses of this joint in particular (thumb) subjected to greater stress than the other joints tends to release the retaining grain in a short time. it would probably help if the grains of the fingers were at least M2 in order to guarantee a better seal against the tendon. please let me know.

pattacini, 2021-09-29:
Nice report @Uboldi80 🚀

Uboldi80, 2021-09-29:
THKS @pattacini !!

closed 2021-09-30 by Uboldi80