robotology/mesh-support #1199: Hands calibration fails - iCubGenova09

davidetome, 2021-09-21:
### Device name 🤖

iCubGenova09

### Request/Failure description

The fingers remain closed during the calibration on both hands.
Logged only the left hand since the right has the thumb oppose tendon broken.

### Detailed context

## video (left hand)
https://user-images.githubusercontent.com/6638215/134181991-1bab2237-0835-4658-8a25-d48e658ba6ba.mp4

## log
[log.txt](https://github.com/robotology/icub-tech-support/files/7203523/log.txt)

Looking at the log seems to be related to the ring and little fingers (j15)

davidetome, 2021-09-23:
The right-hand [thumb oppose](https://github.com/robotology/icub-tech-support/issues/1198) tendon has been replaced (thanks @GiacomoVassallo13 ) but It oscillates

https://user-images.githubusercontent.com/6638215/134534175-6bf40a85-0075-4545-91c8-49bb396fc5e9.mp4

The left-hand j15 randomly fails to calibrate going to push on the palm, it happens more often if other joint are included in the CALIB_ORDER param (may we check motor encoder wires?)

https://user-images.githubusercontent.com/6638215/134534436-dd4e062d-82d1-47f8-a92d-8a9068ab6ae9.mp4

cc @pattacini @maggia80

pattacini, 2021-09-23:
>The right-hand thumb oppose tendon has been replaced (thanks @GiacomoVassallo13 ) but It oscillates

As the tendon has been most likely tightened up, the system has become more reactive. Thus, it could help to reduce the [PID gains][gains] of that joint.

We can only guess some good initial values and then fine-tune them.

```diff
- <param name="kp">                    -500        500         -500        500         </param>
+ <param name="kp">                   -300        500         -500        500         </param>
  <param name="kd">                         0             0              0           0          </param>
- <param name="ki">                       -50          50           -50         50          </param>
+ <param name="ki">                      -40          50           -50         50          </param>
```

[gains]: https://github.com/robotology/robots-configuration/blob/master/iCubGenova09/hardware/motorControl/right_arm-eb28-j8_11-mc.xml#L51-L53

davidetome, 2021-09-24:
# Update

## Left hand
Today, after a deeper investigation, I found the problem.
I noticed that even if the little finger was kept closed, moving the wrist the measure was changing

https://user-images.githubusercontent.com/6638215/134676624-533c8301-fb8a-4b88-ac0e-315936d55b2d.mp4

|            |                 |
|----------|:-------------:|
| ![uno] |  ![due] |

[uno]: https://user-images.githubusercontent.com/6638215/134677007-8974fc10-6fc3-4a99-b755-ac562d9c5909.jpg
[due]: https://user-images.githubusercontent.com/6638215/134677013-9e31536c-2925-4574-aed1-9752e6a81fd6.jpg

I connected the `USB-CAN` adapter to the `mais` board and checked the hall sensors of the little ring.
I finally noticed that the hall sensor in the position highlighted below is no more working

[image]

So we need the help of @gandoo to check and fix it 🔧 

## Right hand
@GiacomoVassallo13 could you please double-check 🔍  the tendon of the right thumb oppose? As you know it oscillates and today w/ @pattacini we noticed that it's looser than the left one. 
This may cause a lack of control.

cc @pattacini @Fabrizio69 @S-Dafarra

GiacomoVassallo13, 2021-09-27:
The cable of thumb oppose joint it was been tensioned, and the worn set screws have been sobstituted.
Ticket: https://github.com/icub-tech-iit/tickets/issues/1683
cc: @davidetome @Uboldi80 @pattacini @Fabrizio69

davidetome, 2021-09-27:
The hands calibrate correctly again.
Thank you @GiacomoVassallo13 @Uboldi80 🎖️ 

https://user-images.githubusercontent.com/6638215/134931999-2c6f987a-5782-48fb-b372-275195fe9457.mp4

https://user-images.githubusercontent.com/6638215/134932447-f52c045c-9fb0-4b3d-b1bc-7f2b29714608.mp4

cc @pattacini @S-Dafarra @Fabrizio69

closed 2021-09-27 by davidetome

pattacini, 2021-09-27:
Cool! Thanks all guys!
If not already done, we should mount back the covers as AMI has the need for shooting a video on Wednesday.

S-Dafarra, 2021-09-27:
> Cool! Thanks all guys!
> If not already done, we should mount back the covers as AMI has the need for shooting a video on Wednesday.
Rather than a video, H&S need to assess the risk while using the robot. Hence we need the robot in a "normal" status. This is going to be on Friday 1/10 afternoon

davidetome, 2021-09-28:
@pattacini do we mount the new tissue on the upper arms covers?

pattacini, 2021-09-28:
If they're ok as they seem, yep 👍🏻

davidetome, 2021-09-28:
> If they're ok as they seem, yep 👍🏻

Maybe @maggia80 wants to take a look before? We can mount the old tissue in the meanwhile since it is very simple to replace

pattacini, 2021-09-28:
> We can mount the old tissue in the meanwhile since it is very simple to replace

Ok 👍🏻

davidetome, 2021-09-28:
@maggia80 approved the new tissue.
Covers mounted by @Uboldi80 🎖️ and new tissue applied.

@salvi-mattia , I left the covers provided by the supplier on your desk. 

cc @pattacini