robotology/mesh-support #1200: iCubGenova09 - Right camera not working

valegagge, 2021-09-23:
### Device name 🤖

iCubGenova09

### Request/Failure description

When I opened the camera device and connected it to Yarpview, the left camera was ok, while the right started to blink and after few seconds the image was completely back.

### Detailed context

I'm sure that this isn't due to a bad Yarp connection, because Yarpview still received images, but they are black.

S-Dafarra, 2021-09-27:
> I'm sure that this isn't due to a bad Yarp connection, because Yarpview still received images, but they are black.

I think it could be the flat cable of the camera. We had similar issues with iCubGenova04 https://github.com/robotology/icub-tech-support/issues/1191

AlexAntn, 2021-09-29:
> > I'm sure that this isn't due to a bad Yarp connection, because Yarpview still received images, but they are black.
> 
> I think it could be the flat cable of the camera. We had similar issues with iCubGenova04 #1191

This issue actually goes further back, since we tried testing things first on Wednesday, and we had several issues with the cameras then. Initially we had the same issue that @valegagge mentioned, we tried power cycling the robot, and after that the cameras no longer worked. This was the log we got from trying to launch the cameras then:

```
yarpdev --from camera/ServerGrabberDualDragon.ini --split true
[WARNING] |yarp.device.grabberDual| Period parameter not found, using default of 0.03 s
[INFO] |yarp.os.Port| Port /icub/cam/left/rpc active at tcp://10.0.0.2:10006/
[INFO] |yarp.os.Port| Port /icub/cam/left active at tcp://10.0.0.2:10007/
[INFO] |yarp.os.Port| Port /icub/cam/right/rpc active at tcp://10.0.0.2:10008/
[INFO] |yarp.os.Port| Port /icub/cam/right active at tcp://10.0.0.2:10009/
[DEBUG] Format = 10
[ERROR] no active cameras
[ERROR] LINE: 336
DragonflyDeviceDriver2: can't open camera
[ERROR] |yarp.dev.PolyDriver| Driver <dragonfly2> was found but could not open
[DEBUG] Format = 10
[ERROR] no active cameras
[ERROR] LINE: 336
DragonflyDeviceDriver2: can't open camera
[ERROR] |yarp.dev.PolyDriver| Driver <dragonfly2> was found but could not open
[ERROR] |yarp.device.grabberDual| Opening devices... FAILED
[ERROR] |yarp.device.grabberDual| Error while opening subdevice
[ERROR] |yarp.dev.PolyDriver| Driver <grabberDual> was found but could not open
[DEBUG] |yarp.dev.Drivers| ===============================================================
[DEBUG] |yarp.dev.Drivers| == Options checked by device:
[DEBUG] |yarp.dev.Drivers| ==
[DEBUG] |yarp.dev.Drivers| device=grabberDual
[DEBUG] |yarp.dev.Drivers| ==
[DEBUG] |yarp.dev.Drivers| ===============================================================
[ERROR] |yarp.dev.Drivers| yarpdev: ***ERROR*** device not available.
[INFO] |yarp.dev.Drivers| Suggestions:
[INFO] |yarp.dev.Drivers| + Do "yarpdev --list" to see list of supported devices.
```

We also thought the issue could be with a flat cable (at least for the image flickering), and in fact, we tried replacing the flat cable on the right eye (me and @GiacomoVassallo13 ), and it didn't change anything. It could be that also this cable is somehow broken, but I would say it is unlikely. Also, given the errors we got Wednesday, it would seem that something was causing the camera device to not start up properly, which would not be the result of a faulty flat cable.

S-Dafarra, 2021-09-29:
Thanks @AlexAntn. I just tried running ``yarpdev`` but I did not get those errors. I was able to correctly see from the left eye, while the right was all black.

Uboldi80, 2021-09-29:
flat cable replaced

the older was demaged.

closed 2021-09-29 by Uboldi80