robotology/mesh-support #1201: Hardware error for left shoulder roll encoder

cedricgoubard, 2021-09-25:
### Device name 🤖

iCubLondon01

### Request/Failure description

Hello,
I get an `[ERROR] sharedcan [1] board 1 ABS ENCODER ERROR AXIS 1` when running `yarprobotinterface` in calibration mode. If I ignore it and run `yarpmotorgui`, I can see that the left shoulder roll has a hardware fault.

### Detailed context

This happened just after I rewired the whole shoulder, so I guess the cause is obvious :-)
However, I have a few questions:
 - Where is the encoder in question? is it the one on the front of the shoulder (see picture below)? If so, I opened it and it does not appear broken... Is there any other way to check?
 - I noticed that there seem to be a missing "screw" on this side of the shoulder compared to the other. Could this have anything to do with the issue?

### Additional context

The `yarpmotorgui` interface:
[image]

The encoder I am mentioning in my question (is it the roll encoder?):
[image]

The missing "screw" on the left side, compared to the right (in the middle of the shoulder):

[image]
[image]

Uboldi80, 2021-09-27:
Hy Cedric 
Yes, the encoder that you mentioned is the ROLL.
Now, probably the problem is caused by a torn thread.
1 point. you must check the AEA encoder :
[image]

2 point: you need to check the chain of encoder:
as you can see, trough the AEA roll and the EMS4 1B0 you have a sectioning. The sectioning is delicated and you should find that in the shoulder. The position of the connector in the BLL is P14.
[image]

let me know

cedricgoubard, 2021-09-27:
Hi @Uboldi80,
Thank you for your answer. I checked the AEA encoder, and I don't see anything wrong:

[image]

I then looked at the EMS4 1B0, but I cannot find the P14 connector. Am I missing something?
I noticed that P8 is supposed to be empty, but is not on our board. Maybe that is where the P14 is connected?
[image]
[image]

Uboldi80, 2021-09-27:
hy

under the P9 connector you should find the P14 (to the other side of the same board)

cedricgoubard, 2021-09-27:
Hi @Uboldi80,
I did not find anything under the board:
[image]

cedricgoubard, 2021-09-27:
I just checked, and the encoder cable is actually connected to P9. One of the wires seems broken; I'll try to fix it and see if that solves the issue.

Uboldi80, 2021-09-27:
ok, great!
Sorry, i thinked your setup was a little bit different in thath board, but aniway, we find the torn wire.

cedricgoubard, 2021-09-27:
No problem, even though the connector was wrong, we still found the issue!
I fixed the wire and it's working again, thank you for your help.

closed 2021-09-27 by cedricgoubard