robotology/mesh-support #1203: Right thumb oppose tendon broken

S-Dafarra, 2021-09-27:
### Device name 🤖

iCubGenova04

### Request/Failure description

During the demo tech week, the right thumb oppose tendon snapped. The cause was probably a too vigorous handshake.

### Detailed context

As above

Uboldi80, 2021-09-28:
Sorry @S-Dafarra  the tendon broken is the thumb proximal.

S-Dafarra, 2021-09-28:
> Sorry @S-Dafarra the tendon broken is the thumb proximal.

Thanks for noticing. I corrected the title.

Uboldi80, 2021-10-01:
@S-Dafarra 
Today, me and @GiacomoVassallo13  replaced the motor and the gear box with the tendon.
unfortunely the gear box was demaged, and the replacement the new set includes some extra work that we necessarily had to do in the mechanical workshop.
Now the joint should be ok.
please let me know next monday

closed 2021-10-05 by Uboldi80