robotology/mesh-support #1205: Bottom motor LED stopped working

cedricgoubard, 2021-09-27:
### Device name 🤖

iCubLondon01

### Request/Failure description

Hello,
One of the 2 LEDs (the bottom one) that usually light up when turning the "motors" switch on iKart `ON` stopped working, and now `yarprobotinterface` fails. Also, the head `CAN` board is no longer detected.

Do you think those 2 problems are related? What do the 2 lights correspond to?

### Detailed context

Here are the LEDs I mentioned; only the top one is working:
[image]

Uboldi80, 2021-09-29:
Hy Cedric 

the Can line cfw2 (3) could be the line start from head to the legs, look the N3 on the top of the image i shared below:
[image]
and below a picture of head can line:
the can is the line with the colors green-black-white and sometimes you will see also the red wire:
[image]

cedricgoubard, 2021-09-29:
Hi @Uboldi80,
I tested the cable of the head CAN, it seemed to be working fine. 
However, after plugging everything back together, now the pc104 does not start anymore!

More specifically, it stopped answering to `ping`, and nothing happens when I try to turn it on with a VGA screen attached.

Do you have any idea how to troubleshoot that?

Here are a few pictures, in case you spot something I missed:
[image]
[image]
[image]

Uboldi80, 2021-09-30:
Hy Cedric

are you sure that the VGA cable are good?
if you are sure, you can try to remove the usb key in to the head and turn on the cpu with the monitor connected. In this case you can see the Bios?
let me know

Uboldi80, 2021-10-08:
Hy @cedricgoubard 

we will contact you as soon is possible to provide you the information requested.

Uboldi80, 2022-04-11:
Hy @cedricgoubard !

Yes, this is a PS CPU, i can provide to send a new one iit code: 1243.D

andreaderito, 2022-04-19:
Hi @cedricgoubard  I'm about to send you the quotation for the supply of the spare part, it should be ready today (19.04.22), I will transmit via email as usual.
cc @Fabrizio69 @maggia80 @Uboldi80

cedricgoubard, 2022-04-20:
Hi @andreaderito,
Great, thank you! I do not think we have received anything yet, I'll let you know as soon as we do.

andreaderito, 2022-04-20:
Hi @cedricgoubard the draft of the quotation is ready (since yesterday) but still pending Scientific Director's signature (I'm going to make a solicit now, so hopefully in short it should be ready and I may send it via email asap) - best

andreaderito, 2022-04-21:
Hi @cedricgoubard quotation sent via email few mins ago - best
cc @maggia80 @Fabrizio69 @Uboldi80

cedricgoubard, 2022-04-21:
Hi @andreaderito, 
We received the quotation and should be issuing the purchase order shortly.

cedricgoubard, 2022-05-12:
Hi, 

We installed the new power board, and our `pc104` boots again! We are now back to the our issues with the CAN bus. Here is what we have:

[image]

And here is everything we noticed so far, based on [this documentation](https://icub-tech-iit.github.io/documentation/icub_wiring/icub1_x/):
 - We tested the `N0` green/black/white wires from  the `CFW-002` to `0B0`, and the connectivity seems fine
 - Our robot does not seem to have `CFB0`; instead, the `N0` coning from `CFB1` gets plug in `P3` of `0B0`, and then exits from `P10` of the same board. I think this might have something to do with our long-term issue with the left leg not being detected? 
 - Also, `P3` and `P10` of `0B0`  have 6 pins, not 4. We assumed that the 2 unused pins are the ones on each side for shield, as shown in [this image](https://user-images.githubusercontent.com/7140395/29571911-dad03504-875a-11e7-87d0-3fd9a88d48a3.png) from [this issue](https://github.com/robotology/icub-tech-support/issues/343)
 - Also, our `0B1` and `0B2` were switched; we put them back correctly, but it did not solve the issue.
 - [The documentation](https://github.com/icub-tech-iit/electronics-wiring-public/blob/master/icub1/iCub%201.7.1/iCub%201.7.1_upperbody_logic.pdf) seems to indicate that the `N0` wire between `CFB2` and `0B2` should not be connected; however, it was on our robot 
 - The blue and yellow LEDs on the side of `0B2` are acting differently from the ones on `0B1`. `0B1` always has both LEDs ON, but `0B2` either has:
   - the yellow ON and blue OFF when we connect `N0` between `CFB2` and `0B2`
   - the yellow blinking and blue OFF when we disconnect them

Since our `iCub` has been repaired many times over the years, I do not know which of those changes were intentional, and which were not.
Could you please help us understand:
- Are `0B1` and `0B2` interchangeable?
- How can we check if they are working correctly?
- Is there anything else we should check?

Thank you for your help!

Uboldi80, 2022-05-13:
Hy @cedricgoubard 
from this link you can see what you should get from FirmwareUpdater under the various CAN lines.
I would say that there are several problems to be solved.
https://icub-tech-iit.github.io/documentation/icub_firmware/associated-firmware/associated-firmware/#icub-v2
Are 0B1 and 0B2 interchangeable? generally the boards of the same type are interchangeable, obviously paying close attention to the can address of the same.
How can we check if they are working correctly? It is almost certainly not a problem of electronic boards, but of interruption of the CAN lines
Is there anything else we should check? 
Interruptions of the can lines almost always occur when they arrive at the can filters and branch out onto the various boards.
Always check the can lines by lightly pulling the cables to verify that there is actually contact between the cable and the connector crimp.
usually after years (as in your case) the can cables tend to cook right on the connectors.
today I try to verify in which points you could have problems.

cedricgoubard, 2022-05-13:
Hi @Uboldi80,
Thank you for answering so quickly.
We just checked the connectivity again: for each small wire in `N0` (black, green, white), we touched the connector on the `CFW` board and the other end on `0B1`, and on `0B2` (see pictures).

[image]

[image]

Everything seems to be working fine... Let me know if you think of any other check we could do.

Uboldi80, 2022-05-13:
Also check the connector on the side that manages the power supply if I'm not mistaken.
pull lightly with tweezers each of those colored cables on the connector.

Uboldi80, 2022-05-13:
you should verify the can filters:

[image]

cedricgoubard, 2022-05-13:
I checked the two CAN filters`CFB1` and `CFB2`: 

[image]

[image]

As far as I can tell, they are both getting 5V from their respective boards. 

I also tried different connections: (not sure it should change anything, but just in case):
 - `CFW` -> `CFB1` -> `0B2`  same error
 - `CFW` -> `CFB2` -> `0B2`  same error
 - `CFW` -> `CFB1` -> `0B1`  same error
 - `CFW` -> `CFB2` -> `0B1`  same error
 - `CFW` -> `0B2` same error 
 - `CFW` -> `0B2` same error 

[image]

Here are the logs I always get from `FirmwareUpdater`:
```bash
icub@pc104:~$ FirmwareUpdater 
Using default address  10.0.1.104
Using default port  3333
"CFW2" - "0"
"CFW2" - "1"
"CFW2" - "2"
"CFW2" - "3"
"CFW2" - "4"
"CFW2" - "5"
"CFW2" - "6"
"CFW2" - "7"
"CFW2" - "8"
"CFW2" - "9"
ACE_SOCK_Dgram: Cannot assign requested address
ACE_SOCK_Dgram_Bcast::mk_broadcast: setsockopt failed: Operation not supported
ACE_SOCK_Dgram_Bcast: Operation not supported
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
[INFO] |yarp.dev.PolyDriver| Created device <cfw2can>. See C++ class yarp::dev::Cfw2Can for documentation.
FirmwareUpdaterCore::getCanBoardsFromDriver(): No answer received from CAN boards after a successful driver init.
[INFO] |yarp.dev.PolyDriver| Created device <cfw2can>. See C++ class yarp::dev::Cfw2Can for documentation.
FirmwareUpdaterCore::getCanBoardsFromDriver(): No answer received from CAN boards after a successful driver init.
```

Uboldi80, 2022-05-13:
Did you also check this filter on the CAN chain?
check on the entire length of the can chain, therefore, arrival and departure for all 3 boards on the head.

cedricgoubard, 2022-05-13:
I think you forgot to include a picture; I assume you mean `CFB0`?
In our case, I did not find this filter on the robot... It looks to me like `P5` on `CFB1` is connected directly to `P3` on `0B0`. Then,   it looks like another CAN line comes out of `P10` on `0B0` and goes somewhere in the back of the robot.

Uboldi80, 2022-05-13:
Ok, this is the picture of BLL 0B0:
[image]
As you can see we have a can line in position P3 until a can filter CFB0

cedricgoubard, 2022-05-13:
Could you tell me where `CFB0` is supposed to be on the robot? I cannot find it on our iCub.

Uboldi80, 2022-05-13:
follow the CAN cable which starts from the P3 of the BLL and which should be identified as N06.
that cable should flow into a can filter (CFB0).
Since you have the BLL in sight, also check that the FLAT CABLE between the BLL and the BLP is well inserted.

gsisinna, 2022-05-17:
Hi @cedricgoubard,
please can you try to center the connector in this picture?

[image]

From what I see now is aligned on one side, you should leave one pin free on the left side and one on the right side.

Thanks!

cc @Uboldi80

cedricgoubard, 2022-05-18:
Hi @gsisinna,
Thank you for suggesting this; however, the connector is already plugged into the center, it just does not look like it on the picture 😄

Uboldi80, 2022-05-20:
Hy @cedricgoubard 
For further verification of the can lines I suggest you do a test that allows us to make sure that the CFW card on the head is good:

Swap the connector of a can line that certainly works and that therefore appears on the firmwareupdater with a can line that does not appear.
For example, I see that you do not have 3 can lines: CFW 0 - CFW 3 - CFW 4 while you see all the other lines instead.
So you could for example swap the connector of CFW 0 with CFW 5 directly on the head and see if actually the can line does not really work or rather it is not the CFW card that has burnt channels.
Please, after this test put everything back in its place!

cedricgoubard, 2022-05-25:
Hi @Uboldi80,

I think you were right! I switched `N0` with `N5` on the CFW, and got the following result (with the torso unplugged):
[image]

Additionally, I just realised that if I have a screen plugged in the `pc104` when discovering the `CAN` channels, I get the following message for buses 0 and 3: `Timeout while flushing HW0/3`.

So, would it be possible to buy another `CFW` board? Is there any newer replacement?
Also, while we wait for a new one, is it possible to change the CAN bus connections on our current `CFW` to only use the boards we need for grasping (*e.g.* head, left arm, left hand)?

andreaderito, 2022-06-05:
> So, would it be possible to buy another `CFW` board? Is there any newer replacement? Also, while we wait for a new one, is it possible to change the CAN bus connections on our current `CFW` to only use the boards we need for grasping (_e.g._ head, left arm, left hand)?

Hi @Uboldi80 may you kindly indicate here the code of the iterm requested by Cedric so to check possible availability of spare, lead time and so on in view of preparing a quotation for Cedric? Thank you.

Hi @cedricgoubard pls be so kind as to send me an email (andrea.derito@iit.it) so to start IIT's internal process for the preparation of the quotation. Thank you very much in advance for your cooperation.

Uboldi80, 2022-06-08:
Hi @cedric
Regarding the problems that you have currently encountered on your platform, I spoke with my referrals and we agreed that the best solution for everyone would be to send the robot here.
In this way we could guarantee you an effective and adequate support aimed at solving all the problems that still exist.

cc. @maggia80 @Fabrizio69 @andreaderito

cedric, 2022-06-08:
FYI - I think you have the wrong cedric.

sgiraz, 2023-09-06:
Hi @cedricgoubard,

see https://github.com/robotology/icub-tech-support/issues/1171#issuecomment-1708359029

AntonioConsilvio, 2024-05-17:
Hi @cedricgoubard! Closing this issue due this comment:

- https://github.com/robotology/icub-tech-support/issues/1171#issuecomment-2117138175

closed 2024-05-17 by AntonioConsilvio