robotology/mesh-support #1206: iCub doesn't turn on

omardarwishokd2, 2021-09-28:
### Device name 🤖

iCub_Urbana01

### Request/Failure description

We are trying to determine the reason for failure. The iCub was working fine but at the moment, it doesn't turn on.

### Detailed context

After turning on the power sources, we go ahead to flip the "CPU" and "Motors" switches on. Everything seems normal when the "CPU" is turned on, but once we flip the "Motors" switch on, the whole system shuts down and the power sources zero out (I believe that's a safety feature). 

Upon further investigation, we found one of the mics' wires completely disconnected from the mic. We tried to reconnect it, but that didn't fix the issue.

We aren't quite sure why the mic would cause this failure, but we want to check:
- Would the mic's absence in the circuit cause such a failure?
- If not, what else would cause the issue?

Regards,
Omar Darwish
[image]

Uboldi80, 2021-09-29:
Hy Omar

i Dont think so the problem was caused by the headphone, but you can  disconnect the connector from the head.
To verify where is the problem you can try to follow se nex steps:

- turn off the motors
- remove the pants
[image]

- disconnect the 2 connectors that power the legs positioned in the left side and in the right side.
the label of this connectors could be LLB 48V...   RLB48V...   look below:

[image]
 
after this action try to turn on the motors and let me know what appened.

omardarwishokd2, 2021-09-29:
Thank you for your response. We disconnected the connectors per your instructions. Unfortunately that didn't fix the issue and we still weren't able to turn the robot on.

Uboldi80, 2021-09-30:
sorry @omardarwishokd2 
can you show me a video of the startup process?
turn on cpu, turn on the motors.
i would see the result, and please give me the Serial Number marked on the chest (S/N.0...)

omardarwishokd2, 2021-09-30:
Hello,

please see video below for the powering up of the robot.

for the SN, I believe it is 015-15-15-15-15

Uploading IMG_2884.MOV…

omardarwishokd2, 2021-09-30:
https://user-images.githubusercontent.com/91555361/135472618-14e570c5-13cf-430c-b0c1-da153b7154a8.MOV

Uboldi80, 2021-09-30:
ok, and if  you disconnect the power cable from icub and turn on both switch?
i wont to be sure that the problem isn't on the power supply.

omardarwishokd2, 2021-09-30:
Doing that turns on the switches normally. Please see video below 

https://user-images.githubusercontent.com/91555361/135485532-46b161eb-7cd9-4890-8816-ae2c63af0307.MOV

Uboldi80, 2021-10-01:
Hy @omardarwishokd2 
Very probably, you have a short circuit in the upper body and probably the problem is ina BLL or BLP board.
I share with you a schematic  pin out of connector that you have in icub back.
As you can see in page 10 to the right side have the pin out description.
considering that the power connection of the legs had been disconnected to isolate the problem, you can try to check with the tester selection the beeper function (acoustic signal).

1. check all the follow connection:
Ext_Larm_B48V
Ext_Larm_U48V
Ext_Rarm_B48V 
Ext_Rarm_U48V
Ext_Waist_B48V
Ext_Waist_U48V
To check the  board from connector you must touch the pins with the tester tips: exemple to chek the Ext_Larm_U48v you need to touch r and h pins if you dont''each beep the board should be ok.
[image]

[wiring-icub_v.
[image]
**WIRING ICUB V1.3:**
1.3.0.pdf](https://github.com/robotology/icub-tech-support/files/7264872/wiring-icub_v.1.3.0.pdf)

omardarwishokd2, 2021-10-01:
Hey @Uboldi80 ,

I did check the pins you mentioned above, and none of them beeped. 

Regards,
Omar

maggia80, 2021-10-04:
Hey @omardarwishokd2,

Could you please check how much is the current limit in the 12V power supply? It should be around 15A. 

It would be nice to see the power supply while switching on the CPU and the motors. Could you please make another video?

thanks,
Marco

omardarwishokd2, 2021-10-04:
Hey @maggia80 ,

something odd happened this morning and the robot actually turned on. I’m not really sure why it did. I attached the video below anyway. If you may please let me know if you have any comments on that video.

Regards

https://user-images.githubusercontent.com/91555361/135870551-1116cf2e-a96f-497f-a5fd-10bd7194c8e5.mov

,
Omar

omardarwishokd2, 2021-10-04:
Also, I understand this is not the section of it, but it would be greatly appreaciated if you can guide me to the location where I could find live-usb configuration instructions. We had to create a new live usb for the icub due to ssh issues (version: legacy) and ssh works fine now. At the moment, I'm not quite sure what instructions I need to follow to compile the new usb.

maggia80, 2021-10-05:
@omardarwishokd2 here you go: https://icub-tech-iit.github.io/documentation/icub_operating_systems/pc104/the-icub-live-installation/

maggia80, 2021-10-05:
@omardarwishokd2 please press the OVP button on the 12V power supply and check which are the setting of the power supply.

omardarwishokd2, 2021-10-05:
Thank you very much @maggia80 .

please see below video requested 

https://user-images.githubusercontent.com/91555361/136049346-ade1dac1-99ce-4283-b54b-ce6d1c4533d4.MOV

maggia80, 2021-10-05:
the current and voltage settings are ok (13.4V and 12A). There is another setting that is the overvoltage. Press the button circled in red and let me know the voltage setting.

[image]

omardarwishokd2, 2021-10-05:
Please see below:

https://user-images.githubusercontent.com/91555361/136110565-2d91624b-54fe-4449-bd59-8e4e4ce92985.MOV

Uboldi80, 2021-10-06:
ok @omardarwishokd2  now the setup run well?

if the problem returns, take a picture from the power supply  in normal and OVP mode.
Thks

omardarwishokd2, 2021-10-06:
Hey @Uboldi80 

Currently it turns on just fine, but I’m having trouble getting it actually move. As I mentioned, we had to create a new live usb for it. If that issue persisted, I can open up a new thread. As for this thread, the robot turns on fine.

Regards,
Omar Darwish