robotology/mesh-support #1207: iCubGenova09 battery is (probably) dead

S-Dafarra, 2021-09-28:
### Device name 🤖

iCubGenova09

### Request/Failure description

We tried to use the battery, but we did not manage to. We are not sure if it is because the battery is completely dead and in protection, or because we did not use it properly.

### Detailed context

We stopped the power supply, then we tried to play with the switch behind the torso cover. The display and the buttons remained always off. Then, we tried to connect the battery charger (the new one of the iCubGenova04 setup, the one with the display). We set the voltage to 42V and the maximum current to 4A. We tried to connect it to the connector fixed in https://github.com/robotology/icub-tech-support/issues/1194, but when we start the output in the battery charger, no current is flowing. We tried both configurations of the switch.

Hence, either we did not understand how to use the battery and the battery charger, or the battery is dead.

### Additional context

Since we have a new battery, I wonder if it could make sense to directly install the new one.

@maggia80 @pattacini @emilianob80

Fabrizio69, 2021-10-01:
@S-Dafarra the battery is faulty (Andrea and Emiliano have tried to turn it back on), so it will be replaced with a new battery.

cc: @Uboldi80 @maggia80 @DanielePucci

emilianob80, 2021-10-01:
@S-Dafarra  new battery is fully charged and ready to be installed

cc: @Uboldi80 @maggia80 @Fabrizio69 @gandoo

S-Dafarra, 2021-10-01:
Great, thanks!

closed 2021-10-05 by Uboldi80