robotology/mesh-support #1210: When running simple funny things, the neck goes often in hardware fault because of overcurrent

S-Dafarra, 2021-10-01:
### Device name 🤖

iCubGenova04

### Request/Failure description

During the Tech Week demo, the neck of the robot was often going in HF when running the scripted funny things. The error was mentioning a high peak current.

We also tried to enlarge the limit, but this was still happening

### Detailed context

cc @vtikha @maggia80 @pattacini

S-Dafarra, 2021-12-01:
@vtikha did you happen to notice this on other robots? Might it be due to some eccentricity of the mechanism? (@salvi-mattia @pattacini)

pattacini, 2021-12-01:
Not aware of any other occurrences of the problem as I'm no longer playing with the robots that much ☹️

>Might it be due to some eccentricity of the mechanism?

Definitely, non-circular pulleys impair the system and make it struggle somehow, although we haven't yet performed thorough analyses to be able to say anything about the extent. Also, are we sure that the neck pulleys are non-cicular on this robot?

S-Dafarra, 2021-12-01:
> Also, are we sure that the neck pulleys are non-cicular on this robot?

No, it was just a wild guess

pattacini, 2022-02-08:
Just `pinned` the issue with the corresponding label to prevent the bot from marking it again as stale.

Uboldi80, 2022-02-08:
si,  in effetti avevo messo in pinned quasi tutte le issue in support al di fuori di quelle non assegnate a me.
adesso ho messo anche quelle

closed 2023-02-21 by github-actions[bot]