robotology/mesh-support #1212: iCubGenova09 - Eth line broken

isorrentino, 2021-10-05:
### Device name 🤖

iCubGenova09

### Request/Failure description

An Eth line is broken. Unable to communicate with torso and legs boards.

### Detailed context

Here the log:

[log_icub-head_yarprobotinterface_1746_ethernet_cut_from_right_arm.txt](https://github.com/robotology/icub-tech-support/files/7288368/log_icub-head_yarprobotinterface_1746_ethernet_cut_from_right_arm.txt)

cc @maggia80 @S-Dafarra @DanielePucci

DanielePucci, 2021-10-05:
Probably better to drop also an email to @Uboldi80  Gandini and @Fabrizio69 @maggia80 to inform them about the early work probably tomorrow they have to address

Uboldi80, 2021-10-06:
fixed e29 eth, was disconnected in the right elbow joint.

closed 2021-10-06 by Uboldi80