robotology/mesh-support #1220: BCB board not working

S-Dafarra, 2021-10-14:
### Device name 🤖

iCubGenova09

### Request/Failure description

The board controlling the battery of the robot is probably fried.

### Detailed context

Before the Maker Faire demo, the robot produced some smoke. @maggia80 @Gandoo @MrAndrea and @Fabrizio69 checked the robot and found some broken components on the BAT board. There was also a broken fuse on the BCB board. While replacing it, some sparks occurred and the BCB stopped working.

Fabrizio69, 2021-10-19:
Hi @S-Dafarra we are working to replace the BMS (ordered a new board)

cc: @MrAndrea

S-Dafarra, 2021-10-19:
> Hi @S-Dafarra we are working to replace the BMS (ordered a new board)
> 
> cc: @MrAndrea @Uboldi80

Do you have any indication about the possible time of arrival? We are going to shoot a video with the robot on the first week of November, and we would need to use the battery

maggia80, 2021-10-20:
BMSs have been delivered to ELTEC for repairing them. As soon they check them they can tell us when the boards will be ready.

S-Dafarra, 2021-11-10:
We received the BMS https://github.com/icub-tech-iit/purchases/issues/1402 https://github.com/icub-tech-iit/purchases/issues/1431, but it still needs to be installed. We have yet another demo the 25 with a TV crew and it is important to have it

Uboldi80, 2021-11-18:
The BMS was replaced and tested.

closed 2021-11-18 by Uboldi80