robotology/mesh-support #1233: iCub 3 neck loosing mechanics

DanielePucci, 2021-11-04:
### Device name 🤖

iCubGenova09 

### Request/Failure description

The neck moves when it should not when blocked. The issue is similar to https://github.com/robotology/icub-tech-support/issues/932

### Detailed context

https://user-images.githubusercontent.com/5045846/140335679-c640a89c-b9be-4618-8e15-6361acbd29f4.MOV

DanielePucci, 2021-11-04:
As agreed with @S-Dafarra, tomorrow early morning, November 5th, would be great to get the attempt to fix the issue

S-Dafarra, 2021-11-10:
@Uboldi80 @gsisinna @GiacomoVassallo13 I think you already did some fixes, but it is still necessary to unmount the mechanism. Am I correct?

GiacomoVassallo13, 2021-11-16:
Buongiorno, la testa di iCub 3 è stata staccata e il collo sottoposto a manutenzione.
Per quanto riguarda il gioco sul giunto dello yau, si è risolto aggiungedo un rasamento trovato  magazzino, per tenere fermo nella sede il cuscinetto.
[image]

Ticket materiale: https://github.com/icub-tech-iit/tickets/issues/1829

GiacomoVassallo13, 2021-11-16:
Purtroppo è stata riscontrata un anomalia grave tra l assieme RC_IIT_010_A_004. 
Il componente RC_IIT_010_P_024 (giallo) risulta storto una volta montato sul pezzo RC_IIT_010_P_008 (grigio).
Immagini della posizione corretta presa a CAD.

[image]
[image]

Immagini dell assieme fisico.
[image]

[image]

cc: @maggia80 @fbiggi @Fabrizio69 @Uboldi80

Uboldi80, 2021-11-18:
the bearings of the neck was replaced and the harmonic inside was blocked trough loctite.
now the neck work well.

closed 2021-11-18 by Uboldi80