robotology/mesh-support #1234: Right shoulder roll optical encoder connector is broken

S-Dafarra, 2021-11-10:
### Device name 🤖

iCubGenova09

### Request/Failure description

The motor encoder of the right shoulder roll is not working well. The joint goes in HF because of i2t error and we are not able to move it from the motorgui. I guess that the connector is broken

### Detailed context

We were doing some wide movements with the arm during the Venice video and probably some of the cables of the connector got damaged.

S-Dafarra, 2021-11-10:
@gsisinna @GiacomoVassallo13

Gandoo, 2021-11-16:
The problem would seem to be the black wire disconnected on the left shoulder connector, I re-crimped the wires and replaced the connector.
Checked if the RE1 cable was intact by disconnecting it to the 2Foc board to check the operation and reading of the encoder.
With the jogger reads.

Uboldi80, 2021-11-18:
The right shoulder connector of encoder was fixed by @emilianob80.
now is ok.

closed 2021-11-18 by Uboldi80