robotology/mesh-support #1235: Tighten the shoulder cables

gabrielenava, 2021-11-11:
### Device name 🤖

iRonCub (iCubGenova04)

### Request/Failure description

The shoulder joints  cables seem loose, and it would be better to tighten them.

### Detailed context

I have noted that after several experiments the shoulder joints become less stiff, and the position control struggles to maintain the desired joint position. Oscillations arise each time one touch the joints. I think the reason is that the shoulder cables become a bit loose, therefore I would like to check and tighten all of them to avoid the shoulder joint to break.

maggia80, 2021-11-12:
@gabrielenava I suggest you follow this documentation: https://icub-tech-iit.github.io/documentation/icub_tendons/shoulder_elbow/ 

let us know if you get any problems.

best,
Marco

GiacomoVassallo13, 2021-11-12:
The shoulder cables have been tensioned. There were loose screws that wobbled the shoulder and elbow joints; I proceeded to tighten them.
I proceed with the closure of the issue.

cc: @maggia80 @Uboldi80 @Fabrizio69

closed 2021-11-12 by GiacomoVassallo13