robotology/mesh-support #1237: Torso cable exited from the guide

gabrielenava, 2021-11-18:
### Device name 🤖

iRonCub (iCubGenova04)

### Request/Failure description

Torso cable on the left side exited from the guide. They do not seem broken, they just slip out the guide.

### Detailed context

The robot was trying to take off, then it fell pitching in the front. To react to this fall, the robot put both arms horizontal and  increased the arm's thrust, but it over reacted and the robot was blown away backward. When it landed back, the cable exited the guide.

Fabrizio69, 2021-11-19:
Hi @gabrielenava the solution of this issue is planned for next Monday 22 with estimated duration: 1 day

cc: @fbiggi @emilianob80 @Uboldi80 @gsisinna

gabrielenava, 2021-11-19:
Perfect, thanks!!

emilianob80, 2021-11-22:
@gabrielenava Fortunately, the repair of the mechanical cable of the torso (cable n ° 11) was faster than expected. The robot is ready to be tested. I close the issue

cc: @Fabrizio69 @gsisinna @Uboldi80

closed 2021-11-22 by emilianob80