robotology/mesh-support #1239: Issues with iCubGenova01 PC104 boot

xEnVrE, 2021-11-19:
### Device name 🤖

iCubGenova01

### Request/Failure description

This evening we had issues while trying to boot up the iCubGenova01 PC104 machine. As discussed with @Uboldi80 I am opening an issue to report the outcome.

Together with @Uboldi80 @gsisinna and @Arya07 we
1. first checked the Ethernet connection by testing a different Ethernet cable attached directly to the PC104
2. then attached a VGA display to the PC104, using the adapter, and verified that the iCub OS was not completing the boot process due to I/O errors on the USB drive device

As suggested by @mbrunettini we downloaded the iCub OS image and dumped it over a new USB driver, updated the .bashrc within and we were able to make the robot starting again.

### Detailed context

This evening we had issues while trying to boot up the iCubGenova01 PC104 machine. As discussed with @Uboldi80 I am opening an issue to report the outcome.

Together with @Uboldi80 @gsisinna and @Arya07 we
1. first checked the Ethernet connection by testing a different Ethernet cable attached directly to the PC104
2. then attached a VGA display to the PC104, using the adapter, and verified that the iCub OS was not completing the boot process due to I/O errors on the USB drive device

As suggested by @mbrunettini we downloaded the iCub OS image and dumped it over a new USB drive, updated the .bashrc within and we were able to make the robot start again.

maggia80, 2021-11-20:
do you think we can suggest following this step to https://github.com/robotology/icub-tech-support/issues/1205 ?

Uboldi80, 2021-11-24:
Hy @maggia80 , we tried to connect the pc 104 of London01 to the vga monito but the pc 104 didn't started

Uboldi80, 2021-11-24:
> Hy @maggia80 , we tried to connect the pc 104 of London01 to the vga monito but the pc 104 didn't started

https://github.com/robotology/icub-tech-support/issues/1205#issuecomment-930296168

gsisinna, 2021-11-24:
After replacing the tendon as reported in https://github.com/robotology/icub-tech-support/issues/1236 , the system had the problem as described in detail by @xEnVrE . The USB flash drive seems to have become corrupted and it was necessary to create a new one using USB 2.0 version. The robot has been brought back to an optimal working condition but it is necessary to replace the USB flash drive in the near future with one provided by the warehouse and of a suitable size, as the current one has a capacity of about 16GB.

After heavy weekend use, the robot seems stable.

cc @Uboldi80

closed 2021-11-24 by gsisinna