robotology/mesh-support #1241: Left hand wrist cable broken

lunagava, 2021-11-20:
### Device name 🤖

iCub[Genova][02]

### Request/Failure description

During issue #1188 with @GiacomoVassallo13, we observed that probably also a left wrist cable is broken. 

### Detailed context

The wrist seems to be not controlled during calibration phase.

Uboldi80, 2021-11-24:
yesterday we tried to replace the wrist tendon (number 10) iit code 6419 that was broken, but the wrist doesn't work well.

i think we could have some problem with the trasmission of this joint.
next week i think, we can try to check what is the problem.

fbiggi, 2021-12-01:
the distal tendon has been repaired

@Uboldi80

gsisinna, 2021-12-02:
Today during the calibration phase I found a problem with the left wrist. It sounds like a motor idling and should be better inspected to identify the problem. The joints in question are JOINTS 5 and 6 of the LEFT ARM (wrist). 

The RIGHT ARM worked and it was calibrated without the aid of delta.

@fbiggi @lunagava

gsisinna, 2021-12-02:
I disassembled the arm and tested the motor in question with the jogger.  As I thought the motor is not turning properly and the cable and pulley need to be checked. Below is the wiring diagram with the motor highlighted: 

[image]

**Motor 1B2M1**

@fbiggi

Uboldi80, 2021-12-03:
Finally i have find the problem in the wrist joint
the motr shaft has broken.
for this reason when the joint is moved passively, the pulley moves and the kinematics is respected, since the passive movement of the joint also moves the winding pulley mounted on the head of the motor.
however, when the motor is moved electrically, since the motor shaft is broken, the motor runs idle.

Unfortunately this problem was quite difficult to find, as sometimes the crankshaft meshed with respect to the pulley, so it seemed to be working.
[image]

gsisinna, 2021-12-06:
Despite the motor intervention there seem to be further problems with joints 5 and 6 of the left arm. The robot was started excluding them from the calibration, but further analysis is needed. 

In particular it is necessary to check the compatibility of the calibration in use (calibration type 3) with the motor that has been used for the replacement, the home position setpoint and the PID marks.

Also below is the excel calibration file used for the old joints: 

[iCub_Calibration_Genova02.xls](https://github.com/robotology/icub-tech-support/files/7660316/iCub_Calibration_Genova02.xls)

gsisinna, 2021-12-14:
Today me and @Uboldi80 investigated the wrist related issue more and we found an incorrect AEA reading for the left one. We then replaced the left wrist AEA chip and fixed the connectors on the MC4-PLUS. Once the correct reading was obtained the wrist calibration was completed with the new values and now both wrists are working.

closed 2021-12-14 by gsisinna