robotology/mesh-support #1247: iCubGenova01 - Left thumb proximal is very loose

xEnVrE, 2021-11-26:
### Device name 🤖

iCubGenova01

### Request/Failure description

The left thumb proximal is very loose and it is thereby difficult to perform grasp actions as the thumb cannot counteract the force exerted by the index and middle fingers.

We verified together with @Uboldi80 that we might need to tighten it a bit.

### Detailed context

The left thumb proximal is very loose and it is thereby difficult to perform grasp actions as the thumb cannot counteract the force exerted by the index and middle fingers.

We verified together with @Uboldi80 that we might need to tighten it a bit.

Uboldi80, 2021-11-26:
me and @gsisinna fixed the tendon and we tested the robot.

closed 2021-11-26 by Uboldi80

xEnVrE, 2021-11-26:
Thank you so much @Uboldi80 @gsisinna