robotology/mesh-support #1248: Left_leg_calibrator timeout while reaching zero position

rustlluk, 2021-11-26:
### Device name 🤖

iCubPrague01

### Request/Failure description

Left leg cant be calibrated and not moving properly.
@matejhof @0xBEDA

### Detailed context

Robotinterface cant calibrate left leg:
Left_Leg_Calibrator : checkGoneToZeroThreshold: joint  1  Timeout while going to zero!
Left_Leg_Calibrator : set 1 : some axis got timeout while reaching zero position... disabling this set of axes

We get the same error even after restarting the robot. If we run the leg from motor gui, the l_hip_roll (Joint 1), do not move at all. The other joint are moving, but slowly even when the speed is increased. But no joint is in error, all of them are "green".

gsisinna, 2021-11-26:
Hi Matej,
the reason could be that the joints are new and have not yet cycled sufficiently. I would suggest you to try to raise the **StartupMaxPwm** from the left leg calibrator file. This is located in the configuration folder for the robot, under the **calibrator** directory, as you can see from the online repo:

[Left_leg_calib.xml](https://github.com/robotology/robots-configuration/blob/master/iCubPrague01/calibrators/left_leg-calib.xml)

In particular at line 33 I would try a startup value of 2000 for each joint. The line will then be as follows:

`<param name="startupMaxPwm">             2000      2000     2000    2000    2000    2000    </param>
`

Let us know if you can move the joints in question this way. For any need we remain at your disposal.

Thank you!

cc @Uboldi80

gsisinna, 2021-11-30:
Any news @matejhof?

rustlluk, 2021-11-30:
We are talking about it on the call right now
[leg.log](https://github.com/robotology/icub-tech-support/files/7624915/leg.log)

rustlluk, 2021-11-30:
We changed the values to 2000, but the leg is till not moving.

gsisinna, 2021-11-30:
@rustlluk can you send a screenshot of your motorgui interface? Thanks

rustlluk, 2021-11-30:
[image]
This is a situation after a manual movement. After the startup all joints of the left leg are in idle

gsisinna, 2021-11-30:
> We changed the values to 2000, but the leg is till not moving.

Try to use a value of 2500 and check if the joints start in idle or not.

davidetome, 2021-11-30:
Looking at the log, the joints failing the claibration are hip roll (j1) and ankle roll (j5)

rustlluk, 2021-11-30:
Ok, changing to 2500 seemed to help. Thanks!
Should we keep this values for now, or we can change it back after some time when the motors get more used?

gsisinna, 2021-11-30:
> Ok, changing to 2500 seemed to help. Thanks! Should we keep this values for now, or we can change it back after some time when the motors get more used?

Try using these values for now and keep me updated on the behavior. After the initial break-in we might try going back to 2000👍

closed 2021-11-30 by gsisinna