robotology/mesh-support #1249: iCubGenova01 - right hand does not calibrate

xEnVrE, 2021-11-26:
### Device name 🤖

iCubGenova01

### Request/Failure description

The right hand does not calibrate, i.e. the robot calibrates all the joints of the right arm except from the 7-th onwards. After the startup these joints are in hardware fault.

Together with @Uboldi80 we have noticed that during the calibration the LED showing the CAN comm. activity switches off indicating that this might depend on the cable connecting the associated board with the robot.

### Detailed context

The right hand does not calibrate, i.e. the robot calibrates all the joints of the right arm except from the 7-th onwards. After the startup these joints are in hardware fault.

Together with @Uboldi80 we have noticed that during the calibration the LED showing the CAN comm. activity switches off indicating that this might depend on the cable connecting the associated board with the robot.

Uboldi80, 2021-11-26:
I have restored the connector of the CAN line directed to the MAI.
we tested the setup and it works.
to be safe I put some glue on the connector itself.

closed 2021-11-26 by Uboldi80

xEnVrE, 2021-11-26:
Thank you @Uboldi80