robotology/mesh-support #1250: Right thumb oppose tendon broken

S-Dafarra, 2021-11-30:
### Device name 🤖

iCubGenova09

### Request/Failure description

The right thumb oppose tendon broke by itself while moving the joint with the motorgui.

### Detailed context

We were doing some tests with @gsisinna, trying to reduce the tension to some of the tendons to try to reduce the oscillations on that joint after the fix of https://github.com/robotology/icub-tech-support/issues/1245. While doing some simple movements on the joint at very low speed, the joint stopped moving and after a second the tendon broke.

cc @GiacomoVassallo13

GiacomoVassallo13, 2021-12-03:
The right thumb oppose joint was been replaced.
Ticket: https://github.com/icub-tech-iit/tickets/issues/1876
cc: @Uboldi80 @gsisinna @Fabrizio69

Uboldi80, 2021-12-14:
Today, me and @gsisinna verify the calibration of the right thumb.
we verify the joint by the GUI, and we noticed that work well.

closed 2021-12-14 by Uboldi80