robotology/mesh-support #1252: front lidar does not work

randaz81, 2021-12-01:
### Device name 🤖

R1SN003

### Request/Failure description

We need to disassemble R1 mobile base to understand why the front lidar does not work anymore

### Detailed context

The motor spins, but the laser does not emit any light. 
If the device is damaged, we can replace it with a spare one (currently on my desk).

Fabrizio69, 2021-12-09:
Hi @randaz81 we'll try to fix this issue next monday (December 13, 2021), when @fbiggi and @emilianob80 will be available for this activity

randaz81, 2021-12-10:
You can find the replacement lidar on the desk near R1's laptops.

emilianob80, 2021-12-13:
@randaz81 R1 has been disassembled and the laser is now available for bench testing. Unfortunately I am not able to do this so someone needs to give us support. If it doesn't work, I will replace it

cc: @Fabrizio69 @fbiggi

emilianob80, 2021-12-13:
As agreed f2f with @randaz81 , I replaced the front lidar and relative electronic control board. The robot now needs to be tested before it is closed

emilianob80, 2021-12-13:
@randaz81 the laser works correctly. @fbiggi closes the robot

cc: @Fabrizio69 @Uboldi80

Fabrizio69, 2021-12-13:
Hi @randaz81, R1 is tested and delivered (covered), can we close this issue?

Uboldi80, 2021-12-15:
the setup was tested and work well.

closed 2021-12-15 by Uboldi80