robotology/mesh-support #1254: iCubLyon/Dijon power connection and startup

pfdominey, 2021-12-07:
### Device name 🤖

iCubLyon01

### Request/Failure description

iCub has just returned from getting fixed up in Genoa.  We no longer use the iKart, and so the power/internet/redbutton/greenswitches connections are different.  Just want to find out the correct connection procedure before startup.

### Detailed context

I attach a photo of the iCub external power connector, and the two power supplies in the rack.
Will I need to take the power supplies out of the rack and connect the power cables?
Please let me know, and send an photos, guides, videos etc.
This will be like the original connection in 2009 before the iKart?
Thanks!  :)
cheers
Peter
[image]
[image]

pfdominey, 2021-12-07:
I also sent an email to Paolo.Uboldi@iit.it related to this.  Thanks.

pfdominey, 2021-12-07:
Woops!  Sorry - I did not look closely enough in the rack box!  I see the additional red switch, green switches etc.  So then do I just connect the fat cable labeled iCubLione directly to the robot and the power supply
[image]
?

gsisinna, 2021-12-17:
We did a videocall this morning to provide the necessary support and get iCubDijon01 up and running at his lab.

cc @Uboldi80

closed 2021-12-17 by gsisinna