robotology/mesh-support #1255: iCubGenova01 - Left thumb moves unexpectedely during yarprobotinterface calibration

Arya07, 2021-12-10:
### Device name 🤖

iCubGenova01

### Request/Failure description

As per title, when launching the yarprobotinterface, during the calibration phase, the left thumb starts doing unexpected movements. 

After a quick check yesterday with @gsisinna, we decided to perform further checks today after the demo.

cc @fedeceola @xEnVrE 

### Detailed context

As per title, when launching the yarprobotinterface, during the calibration phase, the left thumb starts doing unexpected movements. 

After a quick check yesterday with @gsisinna, we decided to perform further checks today after the demo.

gsisinna, 2021-12-17:
Today we have deepened the problem and it is necessary to verify the integrity of the hall sensor for the thumb oppose and the wiring in general of the hand.

Uboldi80, 2021-12-22:
the intervention required to resolve the problem includes a long hardware session.
I replaced the hall sensor, the hall sensor cable that reaches up to the shoulder boards.
Also due to the fact that the magnet was stuck in its seat, today I also replaced the magnet of the thumb oppose.
only the calinration remains to be done.

Uboldi80, 2022-01-04:
Today we replaced the tendon and  improved the joint with M2 granis.

Uboldi80, 2022-01-11:
After trying to calibrate the thumb oppose, we realized that running the Firmware Updater the MC4 of the left arm CFW1 was not displayed.
So I concentrated on repairing the arm can line by replacing at least thirty crimps on various connectors and filters.
After solving this problem, we unfortunately realized that the CAN signal did not reach the MAIS and after an initialization of the card itself the yellow light of the can tended to go out.
to solve the problem I have actually replaced all the crimp of the can line that from the corresponding MC4 reaches the corn passing through the filters.
Despite this long intervention, the problem persisted.
I finally found the cause that tends to knock down the CORN CAN line.
this is a probable short circuit on a thumb hall sensor cable that is not visible to the naked eye.
tomorrow @Gandoo will take care of replacing the cable patch.

Uboldi80, 2022-01-13:
Following the solution of the CAN problem on maize, another problem arose on CFW2 5.
initially it seemed to be the BOARD 1B3 which did not generate the 5v for CAN power supply so I replaced the board with a service spare.
the problem seemed solved, but the left thumb no longer worked.
I later found out that the new card I mounted does not work well.
the problem on the CFW2 5 was again due to an interruption on the CAN line.
solved this problem, we entered the calibration data of the left hand 8 joint.
now the setup seems to work. Today we will close the arm and test the setup.

Uboldi80, 2022-01-13:
after the resolution of the CAN line, I decided to intervene to eliminate various points of failure on the fault line:
[image]
[image]

Uboldi80, 2022-01-13:
We have some problem during the interface:
the wrists have a vibration up to go in fault.
in the logger appares a lot of errors but i have some double:
[log_icub-head_yarprobotinterface_982.zip](https://github.com/robotology/icub-tech-support/files/7862130/log_icub-head_yarprobotinterface_982.zip)

emilianob80, 2022-01-14:
@Arya07 After the maintenance intervention, the left arm was closed. Please test it

[image]

cc: @Uboldi80 @gsisinna

Uboldi80, 2022-01-14:
Thanks @emilianob80 

as soon is possible i will test the robot.

Uboldi80, 2022-01-21:
After @emiliano closed the robot I checked the left arm and now it seems to work correctly.
Unfortunately, several problems appeared on the right arm:
problems on cfw6-cfw2-vibration of the wrist which seems to bring the setup in HF.

Uboldi80, 2022-01-21:
after having restored various power supplies on the CAN and on the MC4, the problem of CAN2 and the pulse problem remained

Uboldi80, 2022-01-21:
after identifying some points on the can2 line including a filter that gave problems, the CFW2 has stopped working completely.
This latent problem had appeared a few times while using the setup.

Uboldi80, 2022-01-21:
Yesterday afternoon after a thorough investigation and with the help of @gandoo we found the malfunction of a BLP SN.045 placed behind the back right side, which brings the CAN2 towards the arm.
replacing the BLP the cfw2 now brings back all the hardware present in the arm by launching FirmwareUpdater.

Uboldi80, 2022-01-21:
Now by throwing only the right arm, the robot configures it, but the problem of vibration on joints 5-6 remains, which does not seem to be solved yet.
Analyzing the raw data again, it can be seen that the reading on joint 5 is not stable, so I proceed to replace the AEA1 code 1810 card placed on the wrist.
  Following the replacement of the 1810, the wrist is stable and the reading is consistent with the position.

Uboldi80, 2022-01-21:
Following this last intervention, the setup has started working again, with the exclusion of the left distal thumb which needs to be recalibrated, and for which I ask Nicola Piga to open a specific issue.
Analyzing the path taken on this intervention that started from the need to restore the Thumb oppose left it is evident that the age of the setup has led to a series of chain faults apparently unrelated to each other but united by the fact that putting the robot's hand the risk that it involves is very high.

closed 2022-01-21 by Uboldi80