robotology/mesh-support #1260: iCubGenova09 r_thumb_proximal and r_thumb_oppose does not work as expected

kouroshD, 2021-12-15:
### Device name 🤖

iCubGenova09 

### Request/Failure description

In practice, there are two problems that I detail them as follows.
-  Given that `r_thumb_proximal` has some fixed values in the middle range (e.g., `40 deg`),  when we give a `r_thumb_oppose` command (to track the requested reference position value in position control mode), there are huge error on `r_thumb_proximal` while `r_thumb_oppose` tries following the requested position.
- When the initial position of `r_thumb_proximal` and `r_thumb_oppose` are low values, it is not possible to control one of the two axis effectively.

The desired behavior is that, in the permissible range of motion provided by each of those two axis, we can effectively control each of them separately.

### Detailed context

The accompanying video show the behavior.

https://user-images.githubusercontent.com/17707730/146179345-af8eff98-aafc-474b-a192-4e86002c4f88.mp4

Uboldi80, 2021-12-24:
I tried to pull the thumb oppose and the proximal but there is a problem with the oppose tendon.
it must be replaced because it is about to break.

Uboldi80, 2022-01-03:
today we replaced an M1.6 grain of the thumb oppose with an M2, after that which @GiacomoVassallo13 and @gsisinna   took care of the replacement of the thumb oppose.

Uboldi80, 2022-01-03:
the thumb proximal was tightenedand semms t work well

gsisinna, 2022-01-04:
> today we replaced an M1.6 grain of the thumb oppose with an M2, after that which @GiacomoVassallo13 and @gsisinna took care of the replacement of the thumb oppose.

Today I replaced the thumb oppose cable and now it works well. I tested it from 0 to 90 degrees.

cc @Uboldi80

closed 2022-01-04 by gsisinna

S-Dafarra, 2022-01-20:
Awesome! For future reference, I guess that the problem was what mentioned in https://github.com/robotology/icub-tech-support/issues/1280#issuecomment-1016181970