robotology/mesh-support #1263: [iCubGenova04] A cable is damaged inside the robot head

gabrielenava, 2021-12-21:
### Device name 🤖

iCubGenova04

### Request/Failure description

We noticed that there is a damaged cable inside the robot head:

[image]

would if be possible to fix/replace it?

### Detailed context

It is not urgent (the robot works fine for now) and the robot is still on the roof, I just opened the issue but working on it on January if fine to me.

cc
@Uboldi80 @Fabrizio69

gabrielenava, 2022-01-17:
hello, would it be possible to schedule this fix in the next days of the week? The robot is currently not connected to the crane and we can bring it down, I think this will simplify a lot the fixing procedures

Uboldi80, 2022-01-17:
could you bring me the setup for Wednesday?

gabrielenava, 2022-01-17:
>could you bring me the setup for Wednesday?

sure!

emilianob80, 2022-01-21:
@gabrielenava The cable was shortened and repaired using self-agglomerating tape and braiding

[image]
[image]

CC: @Uboldi80 @Fabrizio69

closed 2022-01-21 by emilianob80