robotology/mesh-support #1264: The left wrist pulleys are not tight and the cables might slip out of place

S-Dafarra, 2021-12-21:
### Device name 🤖

iCubGenova09

### Request/Failure description

During some intensive tests with some loads on the robot hands, we noticed that one of wrist cables slipped out. @GiacomoVassallo13 noticed that there is some space between the pulleys that should not be there.

@DanielePucci and @salvi-mattia managed somehow to put the cable back in place.

### Detailed context

There is some space in the wrist pulleys.

cc @Fabrizio69

Uboldi80, 2021-12-23:
@GiacomoVassallo13  replaced the tendon and i have done the rist calibration and now the joint work well, but we still have a vibration problem.

Today i try to fix it, maybe, could be a problem with the tensioning of the cable.

Uboldi80, 2021-12-23:
Today i have changed the AEA sensor and the vibration is solved.
now the joint work well

closed 2021-12-23 by Uboldi80