robotology/mesh-support #1266: [iCubErzelli01] Issue with the board torso-eb5-j0_2

ddetommaso, 2021-12-27:
### Device name 🤖

iCubErzelli01

### Request/Failure description

The robot does not even start calibrating. 

### Detailed context

The issue seems related to broken connections with the board of the torso.

### Additional context

[log_icub-head_yarprobotinterface_1528.txt](https://github.com/robotology/icub-tech-support/files/7779595/log_icub-head_yarprobotinterface_1528.txt)

maggia80, 2021-12-29:
@ddetommaso I added Gabriele Sisinna. I think it could an issue related to some disconnection. If you are in the lab could you run the _FirmwareUpdater_ to see which boards are not visible in the ETH network?

ddetommaso, 2021-12-29:
@maggia80 I can confirm that from FirmwareUpdater the board torso-eb5 (10.0.1.5) is not visible.

maggia80, 2021-12-29:
@ddetommaso is the only one that is not visible?
Or also the 10.0.1.6 ... are not visible?

https://icub-tech-iit.github.io/documentation/icub_wiring/icub2_x/#ethernet-backbone

ddetommaso, 2021-12-29:
@maggia80 this is the current situation
[image]

maggia80, 2021-12-30:
[image]

Ok so the issue is in the right arm. None of the cards of the right arm are visible

Uboldi80, 2022-01-05:
We solved the problem with the right arm:

the power connectors were fused
probably the connections between the crimp and the cables had gone loose causing an arc and overheating the connector part.
we therefore replaced the 3 connectors on the MC4 PLUS and redone the power supply wiring.

closed 2022-01-05 by Uboldi80

gsisinna, 2022-01-05:
> We solved the problem with the right arm:
> 
> the power connectors were fused probably the connections between the crimp and the cables had gone loose causing an arc and overheating the connector part. we therefore replaced the 3 connectors on the MC4 PLUS and redone the power supply wiring.

Some pics of the broken connector:
[image]
[image]
[image]