robotology/mesh-support #1269: Untight left wrist

DanielePucci, 2022-01-05:
### Device name 🤖

iCubGenova09 

### Request/Failure description

The left wrist seems not tight, or with cable issues. 

### Detailed context

https://user-images.githubusercontent.com/5045846/148206830-8f06c67f-560c-4e55-8d18-d1527bb9ccb8.MOV

CC
@S-Dafarra @kouroshD

GiacomoVassallo13, 2022-01-05:
Buongiorno, tutti i cavi del polso sono stati tirati e messi a registro. Nello specifivo i cavi dek
l WRIST MOTOR TENDONS e del WRIST - PASSIVE TENDONS. Attendiamo feedback @S-Dafarra.
cc: @fbiggi @DanielePucci

DanielePucci, 2022-01-05:
Un po meglio, ma come da video le vibrazioni sono comunque evidenti @GiacomoVassallo13

fbiggi, 2022-01-05:
In accordo con @S-Dafarra @DanielePucci @Fabrizio69 
quando @Gandoo sostituirá il cabalggio faremo check insieme a @gsisinna sul gruppo polso mano sx

DanielePucci, 2022-01-05:
Si ok @fbiggi, la cosa interessante è che se si bloccano i cavi del polso destro con le dita, il polso non si muove (o di poco), invece se si fa lo stesso con il polso sinistro, il polso si muove.

gsisinna, 2022-01-10:
The left wrist cables have been replaced. An additional check will be done with the guys from the team.

fbiggi, 2022-01-10:
endons 9-10 code replaced
9 6418 51 mm RC_IIT_019_G_027
10 6419 43 mm RC_IIT_019_G_028

I enclose a ticket for the material required for construction 

https://github.com/icub-tech-iit/tickets/issues/1935#issuecomment-1008617795

cc; @Fabrizio69 @maggia80

gsisinna, 2022-01-11:
After initial testing it appears that replacing the wrist tendons is not sufficient. 

The wrist continues to oscillate and go into resonance as soon as it is perturbed from the outside. The cable tensioners have been set up as best as possible and the only suboptimal solution obtained was to reduce the proportional gain of joints 5 and 6 to prevent resonance from occurring.

It is necessary to better investigate the whole mechanical structure of the wrist and understand if there is excessive play between the various mechanical components. 

cc @fbiggi

gsisinna, 2022-01-12:
From the CAD analysis, I have identified a possible screw to tighten to reduce wrist play. This is located under the magnet and is highlighted in the figure:

[image]

Today we will move forward with this intervention.

cc @DanielePucci @Uboldi80

gsisinna, 2022-01-12:
Other details regarding the bearings:

[image]

[image]

gsisinna, 2022-01-12:
After checking the tightness of the above screw, we saw that the play of the components is not dependent on that. After having consulted the team, it was decided by mutual agreement to send the arm to the mechanical workshop and to disassemble everything to solve the problem. 

The left arm was delivered to Marco Pinaffo today.

cc @DanielePucci @fbiggi @Fabrizio69

Uboldi80, 2022-01-14:
Very good 

now after the intervention to solve the wrist, probably we need to recalibrate the joint casue, the magnet position of that encoder was changed.

Uboldi80, 2022-01-17:
Today i calibrate the wrists and i think that the joint work well.
the vibration seems to go away.

DanielePucci, 2022-01-17:
That's great guys, thanks a lot!

DanielePucci, 2022-01-20:
Today we turned on the robot and checked the wrists: it seems that the problem still persists, and the PID gains are the same. Re-opening. We may eventually discuss whether or not to intervene

https://user-images.githubusercontent.com/5045846/150327813-aad058cf-2110-4924-9864-19e08b39094f.mp4

DanielePucci, 2022-01-20:
CC @S-Dafarra @Uboldi80 @gsisinna @Fabrizio69 @maggia80

gsisinna, 2022-01-20:
Last time I achieved resonance avoidance by setting the proportional gain from 400 down. In particular, the asymmetrical behavior between left and right after the intervention makes me think of a problem more in the control and stability of the system, for example in the position feedback given by the sensor.

S-Dafarra, 2022-01-20:
> Last time I achieved resonance avoidance by setting the proportional gain from 400 down. In particular, the asymmetrical behavior between left and right after the intervention makes me think of a [...]

I think the encoder on the wrist has been changed recently. Usually problems in the encoder result in high frequency vibrations. 

Usually these oscillations are caused by mechanical backlash :thinking:

gsisinna, 2022-01-20:
> I think the encoder on the wrist has been changed recently. Usually problems in the encoder result in high frequency vibrations.
>
> Usually these oscillations are caused by mechanical backlash 🤔

Watching better the video it seems also that the oscillation is present only on a degree of freedom that is a part of the ''X'' on which the wrist moves, then considering the low-level decoupling directly from the firmware.

Uboldi80, 2022-01-21:
I think that unfortunately the mechanical play on the parts that was repaired in the workshop has reappeared again.
To restore the functionality of the wrist it will be necessary to replace some mechanical parts of the assembly.
We are working to identify them and to verify if they are available.

Uboldi80, 2022-01-27:
Today, with Marco Pinaffo e @GiacomoVassallo13 fixed the left wrist.
It was necessary to replace some components that obviously showed signs of wear and that perhaps originally could have been mounted in a non-optimal way.
mainly the parts that gave problems were fundamentally 3:
1 WRIST PITCH SHAFT 3544.A2
2 WRIST PITCH PULLEY 3551.A2 
3 PIN 2x14 H8 2361 ------>  REPLACED WITH PIN 2X14 m6 13139

In the screenshot below you can see the assembly of the 3 parts and how they are coupled.

the pin crosses the shaft and enters the pulley forming an integral part.

[image]
Unfortunately we realized that the coupling that these parts have is not ideal for maintaining rigidity over time.
the hole in the shaft is 2H7, the one in the pulley is H7 while the pin is 2 h8.
to improve the coupling we have introduced a 2x14 m6 pin.
in this way the coupling of the pulley shaft by forcing is guaranteed.
the pin crosses the shaft and enters the pulley forming an integral part.

closed 2022-01-27 by Uboldi80