robotology/mesh-support #1272: Some fingertip skins of left and right hands of the iCubGenova09 are broken

kouroshD, 2022-01-07:
### Device name 🤖

iCubGenova09 

### Request/Failure description

Some of the fingertip skins of the robot, both on the left and right hand are broken.

### Detailed context

Following the issue in https://github.com/robotology/icub-tech-support/issues/1261, on the 23rd of December all the skins were working.

After that day, we tried the skins on this Tuesday (5th of Jan), some of them were broken again.

Uboldi80, 2022-01-07:
today with @S-Dafarra checked the fingertip broken:

left hand:  pinkie and index
right hand: thumb

Uboldi80, 2022-01-10:
right thumb fingertip replaced

Uboldi80, 2022-01-11:
i fixed the L0 and L3 fingertips in the left hand.

closed 2022-01-17 by Uboldi80