robotology/mesh-support #1273: Right thumb oppose tendon broken

S-Dafarra, 2022-01-07:
### Device name 🤖

iCubGenova09

### Request/Failure description

During a normal calibration, the tendon snapped on its own.

### Detailed context

The right thumb oppose was moving to the classical position during calibration, and the cable snapped on its own.

cc @pattacini @kouroshD @DanielePucci

gsisinna, 2022-01-10:
The broken cable was the right thumb proximal. This has been replaced and it will be checked again for calibration once the robot is running.

emilianob80, 2022-01-14:
After the tendon was replaced by @gsisinna , the arm was reassembled on the robot and rewired in the missing parts.

@S-Dafarra  can you carry out a functional test when you can?

cc @pattacini @kouroshD @DanielePucci @Uboldi80

Uboldi80, 2022-01-14:
Thanks @emilianob80

S-Dafarra, 2022-01-14:
> @S-Dafarra can you carry out a functional test when you can?

Yes, sure!

Uboldi80, 2022-01-19:
the thumb oppose is ok and seems to work well

closed 2022-01-19 by Uboldi80