robotology/mesh-support #1278: [iCubErzelli01] General maintenance, robot update and other issues

ddetommaso, 2022-01-18:
### Device name 🤖

iCubErzelli01

### Request/Failure description

We would need to update the entire robot (head and boards) and ask for support with the other open issues by the end of February.

### Detailed context

- Complete update of the robot head and boards
- Fix joint 4 calibration drift #1270 
- Fix EB11 (MS4) right leg overheating #1271
- Overall tightening of cables, electrical connections and other generic check up

### Additional context

The robot would be available for maintenance between the 14th to the 28th of February.

ddetommaso, 2022-01-18:
cc @Nicolasr3

gsisinna, 2022-02-16:
Today, together with @Uboldi80 and @Nicogene, the software (icub-head) of iCubErzelli01 has been completely rebuilt with an installation of the latest version of iCubOS and robotology superbuild. It has been made a configuration of the local network for the icub-head and updated the firmware of the boards that were left in previous versions. The yarprobotinterface is now working with the new installation.

It remains to make the calibration and update the files in robots-configuration.

closed 2022-02-17 by gsisinna