robotology/mesh-support #1279: [iCubErzelli03] General maintenance, robot update and other issues

ddetommaso, 2022-01-18:
### Device name 🤖

iCubErzelli03

### Request/Failure description

We would need to update the robot head and ask for support by the end of February.

### Detailed context

- Complete update of the robot
- Overall tightening of cables, electrical connections and other generic check up

### Additional context

The robot would be available for maintenance between the 14th to the 28th of February.

ddetommaso, 2022-01-18:
cc @Nicolasr3

gsisinna, 2022-02-16:
Today we tried to install the latest version of the icub-head software, but we encountered some malfunctions in the video drivers of the BIOS that do not allow the selection of the primary boot device and then to proceed with the installation. 

Tomorrow we will try to update the bios in an alternative way and we will do some tests with different peripherals to discriminate the problem.

cc @Uboldi

Nicogene, 2022-02-17:
Today @gsisinna managed to install ubuntu 20.04 throug the auto install key and after the installation of the latest release of the superbuild we had to change the XML files to run the robot interface.
Here is the PR containing the fixes https://github.com/robotology/robots-configuration/pull/338

gsisinna, 2022-02-17:
For the installation of the updated operating system I had to use GRUB in command line mode and manually mount the USB key because the BIOS is still malfunctioning. 

In the future a check should be done on the video drivers of the BIOS or the connectors responsible for the communication.

Regarding the software update and the installation of robotology superbuild the system is in line with the latest version and the firmware of the boards have been updated including the RFE board.

cc @Uboldi80 @Fabrizio69

closed 2022-02-17 by gsisinna