robotology/mesh-support #1283: There are problems with iCubGenova09 Hands

kouroshD, 2022-01-20:
### Device name 🤖

iCubGenova09 

### Request/Failure description

Today, I did a comprehensive check of the left and right hand of the robot. There are some small problems with the robot hand (compared to previous ones) which blocks us from teleoperating successfully the robot hands. They are mainly with the right hand. 

### Detailed context

Here are the list of the problems:
## Left Hand
1.  `l_index_distal` does not reach `0 deg` and it stops in `14 degs`

## Right Hand
2. When I started the yarprobotinterface, the `r_hand_finger` started in non-configured status.
3. `r_pinky` is not well calibrated. When the robot is in `0 degs` in reality, it shows `102 deg` on motorGui.
4. `r_middle_proximal` often goes to HW fault
5. `r_index_distal` is not calibrated
6. right thumb skin does not work
7. `right-index-middle` **hall sensor** does not stream data (always `0`)
8. `right-pinky-proximal` **hall sensor** does not stream data (always `0`)

cc @DanielePucci @S-Dafarra

Uboldi80, 2022-01-21:
Hy @kouroshD today I will take a look at the problems on the hands, as regards the finger abduction joint 7 right hand, it has been excluded from the calibrator, for this reason you see it not configured, as it has a problem that at the moment we have not been able to solve (i.e. when is in control and the palm is touched the joint changes position)

kouroshD, 2022-01-21:
Thanks @Uboldi80

kouroshD, 2022-01-24:
Hi @Uboldi80 . I talked with @DanielePucci for the use of the right thumb fingertip skin of iCubGenova04 for the iCubGenova09 (due to the lack of materials). He gave OK for that. So please feel free to substitute it as we discussed on Friday. Thanks.

cc. @gabrielenava @S-Dafarra

Uboldi80, 2022-01-24:
today I replaced the R0 fingertip recovering it from Genova04.
Unfortunately it wasn't working.
We are now building a new one (R0) so that we can replace it again.

closed 2022-01-24 by Uboldi80