robotology/mesh-support #1285: Feasibility study for the introduction of RFE Master on iCubGenova04

gabrielenava, 2022-01-21:
### Device name 🤖

iCubGenova04

### Request/Failure description

Feasibility study for the introduction of RFE Master on iCubGenova04

### Detailed context

cc @Uboldi80

Uboldi80, 2022-01-21:
Following the continuous unresolved problems on facial expressions, ( the possibility of introducing the RFE master for the management of facial expressions is evaluated.
The documentation relating to the feasibility study that identifies the parties involved in this upgrade is inserted below.

pattacini, 2022-01-21:
This is not a request for support.
This is more a development ticket.

I'm gonna open up a request into our internal workflow.

closed 2022-01-21 by pattacini