robotology/mesh-support #1288: Issues with R1SN003 left arm

elandini84, 2022-01-27:
### Device name 🤖

R1SN003

### Request/Failure description

The left arm of the robot R1SN003 has some issues. The `yarprobotinterface` throws a set of errors while trying to calibrate the left hand and as a result the left wrist tripod is grayed out in `yarpmotorgui` and the left hand is joints (`l_pam` or JOINT_0 and `l_thumb` or JOINT_1) are in idle by default and if one tries to run them they behave erratically (the palm position slider jumps back and forth and if ones tries to open the palm, it just gets back to a semi-closed position).

### Detailed context

Looking at the log of the `yarprobotinterface` the first error message is:

> [ERROR] Left_Hand_Calibrator : checkGoneToZeroThreshold: joint  0  Timeout while going to zero!

Followed by:

> [ERROR] Left_Hand_Calibrator : set 1 : some axis got timeout while reaching zero position... disabling this set of axes

And then this error:

> [ERROR] unable to complete calibration: joint 1 in 'not_configured' status inside doneRaw() function

is repeated several times (also for joint 2)

Another thing that keeps getting printed is this warning:

> [WARNING]  from BOARD 10.0.1.3 (EB3), src LOCAL, adr 0, time 58s 381m 738u: (code 0x0200000a, par16 0x0003 par64 0x0000000000000002) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .

### Additional context

In this attached file there is the complete log from the `yarprobotinterface`
[yarprobotInterface_Log-Issues_with_R1SN003_left_arm.txt](https://github.com/robotology/icub-tech-support/files/7949916/yarprobotInterface_Log-Issues_with_R1SN003_left_arm.txt)

Uboldi80, 2022-01-28:
We fix the problem on the left arm:

the belt was disconnected by the gear shaft and the gear on the shaft was loose.
now we need to feddback from you @elandini84 

thanks for the intervention @emilianob80 , @fbiggi

elandini84, 2022-01-28:
Thanks a lot.

gsisinna, 2022-01-31:
> Thanks a lot.

Hi @elandini84, 
can we close this issue?

elandini84, 2022-01-31:
> > Thanks a lot.
> 
> Hi @elandini84, 
> can we close this issue?

Sorry for not replying sooner. I have not been in the lab these days. I will be able to test the robot tomorrow morning.

gsisinna, 2022-01-31:
No problem @elandini84, we remain available for any kind of support or problem!

elandini84, 2022-02-01:
@gsisinna I just tested the robot and now everything is fine. So, for me, the issue can be closed. Thanks a lot.

closed 2022-02-01 by Uboldi80