robotology/mesh-support #1289: AEA invalid reading error

S-Dafarra, 2022-01-28:
### Device name 🤖

iCubErzelli02

### Request/Failure description

```
	4268,293279	ERROR			 from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 1348s 604m 526u: (code 0x02000008, par16 0x0002 par64 0x0000000000000009) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
```

### Detailed context

Here the log 
[log_icub-head_yarprobotinterface_1941_AEA_invalid_data.txt](https://github.com/robotology/icub-tech-support/files/7959834/log_icub-head_yarprobotinterface_1941_AEA_invalid_data.txt)

cc @gsisinna

gsisinna, 2022-02-02:
It seems that the problem is not due to incorrect AEA readings. Tomorrow we will try to replace anyway the AEA board and also the STRAIN that could be involved in the hardware failure of the left arm.

As reference see https://github.com/robotology/icub-tech-support/issues/201

S-Dafarra, 2022-02-02:
Note that we were not using torque control at all, hence it is a bit suspicious to me that the joint was going in HF because of the strain.Usually, it is a connector problem. See:
- https://github.com/robotology/icub-tech-support/issues/1164
- https://github.com/robotology/icub-tech-support/issues/1145
- https://github.com/robotology/icub-tech-support/issues/1148

Uboldi80, 2022-02-06:
@emilianob80 please:
Tomorrow morning can you prepare the AEA2 with the complete working from AEA to EMS4.
I would ti replace It.
Tnks! And, can you orepare 1 EMS4 and 1 2foc for the maintenance box in erzelli?

emilianob80, 2022-02-07:
> @emilianob80 please: Tomorrow morning can you prepare the AEA2 with the complete working from AEA to EMS4. I would like to replace It. Tanks! And, can you prepare 1 EMS4 and 1 2foc for the maintenance box in erzelli?

@Uboldi80 The sensor you are asking me cannot be bench wired. The PCB must be soldered in place because the cable passes through a mechanical piece.
As far as programming is concerned, given the lack of electronic cards, it is advisable to bring those already present in the suitcase and program them.

emilianob80, 2022-02-07:
@Uboldi80 cable and boards ready for collection

gsisinna, 2022-02-07:
Here the output of `yarprobotinterface`:

[yarprunlog_07_02_2022_08_48_32.log](https://github.com/robotology/icub-tech-support/files/8013294/yarprunlog_07_02_2022_08_48_32.log)

[log_icub-head_yarprobotinterface_1973.txt](https://github.com/robotology/icub-tech-support/files/8013300/log_icub-head_yarprobotinterface_1973.txt)

In the last interventions, several electronic harnesses have been restored and today we will proceed with the replacement of the AEA provided by @emilianob80 

cc @Fabrizio69

gsisinna, 2022-02-07:
Today the problem also occurred on the wrist, which remains in hardware fault after several boot attempts with the interface:

`	253,590653	ERROR			 from BOARD 10.0.1.24 (left_arm-eb24-j4_7), src LOCAL, adr 0, time 320s 819m 813u: (code 0x02000008, par16 0x0001 par64 0x0000000000000002) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
`

gsisinna, 2022-02-07:
After several tests, multiple latent problems were identified in the left arm electronics. Among them, today the wrist was fixed and two AEA boards (roll and yaw) were replaced. It seems that the problem has been solved and the yoga demo did not lead to hardware faults anymore. 

cc @Uboldi

Uboldi80, 2022-02-07:
https://user-images.githubusercontent.com/80786976/152803471-c12ec78d-1a0a-4593-a254-269c1674f0c7.mp4

closed 2022-02-07 by Uboldi80