robotology/mesh-support #1290: Fix small problem with iCubGenova09 hands

kouroshD, 2022-02-01:
### Device name 🤖

iCubGenova09 

### Request/Failure description

There are some small problems with icub left and right hands to check.

### Detailed context

- Right hand pinky fingertip skin does not work.
- `l-middle-proximal` and  `l-middle-distal` fingers limits each other, so together they cannot reach `0 deg`.
- `l-hand-finger` needs to be opened a bit so that when closing left index and middle they do not collide with each other.
- `r-thumb-distal` at the moment cannot reach `0` degrees. Maybe this is because of the rubber glove on top of the robot thumb. to be checked.
cc @DanielePucci  @S-Dafarra

Uboldi80, 2022-02-08:
today i fixed the soldering in the pinkie.
now should be ok

Uboldi80, 2022-02-28:
Today I checked the distal and proximal middle:
The glove creates some problems in the return of the joints causing some additional friction.
For this reason I have withdrawn the tendons, recalibrated the distal middle and slightly modified the glove on the lower part of the distal.

The same goes for the thumb, which had the glove conflicting with the movement of the joint.

I changed the calibration values and installed them by committing them.
I ask you before pulling to verify them.

Now the movement of the fingers on the right hand seems acceptable to me.

closed 2022-02-28 by Uboldi80