robotology/mesh-support #1291: iCubGenova09 lef wrist and right thumb do not calibrate

S-Dafarra, 2022-02-01:
### Device name 🤖

iCubGenova09

### Request/Failure description

They were calibrating fine until we installed again the robots-configuration files. Then, the calibration fo the right thumb and of the left wrist got lost. I think that the installed files were modified instead of the source ones

cc @Uboldi80 @gsisinna 

### Detailed context

We were modifying the configuration files to increase the joint limits. After installing them, the right thumb and the left wrist did not calibrate anymore.

S-Dafarra, 2022-02-01:
Thanks to the support of @gsisinna and @Uboldi80, we figured out what probably happened.

Yesterday, when performing some experiments on the robot configuration files, the branch with the configuration files changed. Unfortunately, there were some modified files containing the mods fixing the right thumb and the left wrist. These files were not committed and the modifications got somehow lost when switching between the branches. 

When we arrived this morning the working directory was clean, without any modification. The robot was also starting correctly, but as soon as we installed the configuration files again, we started noticing the problems.

Luckily @davidetome saved a backup of the robot configuration files from which we managed to restore the mods. The interesting thing is that the backup was of the ``install`` folder, not of the source.

My guess is that, when switching branches yesterday, the backup folder was used to restore the ``install`` version of the configuration files, but not of the ``source``, so when installing again, the restore ``install`` folder got lost, and with it the necessary modifications.

Then, by using the same backup,  @GiulioRomualdi and @paolo-viceconte managed to restore the lost modifications and commit them with https://github.com/ami-iit/robots-configuration/commit/13c5fb477185fa142eb58879214afa3697f9d9fc

I think that the main problems here were the following:
- those important modifications were not yet pushed, despite being very important
- we have no track of what parameter modification was needed after each intervention
- we did not launch the script to check the differences between the source and the install before installing

At least for the first problem, I have granted @gsisinna and @Uboldi80 with push access to the https://github.com/ami-iit/robots-configuration, where we store the leading edge of the configuration files we use on the iCubGenova09 robot.

For the second problem, my suggestion would be for @Uboldi80 and @gsisinna to add directly in the issue the link to the commit. In this way, it should be easier to recover from similar situations.

For the third problem, I am not sure how to face in some automatic way.

Let me cc also @pattacini in case he wants to suggest other good practices. Closing for the time being, as the problem per se has been solved.

closed 2022-02-01 by S-Dafarra

pattacini, 2022-02-01:
>At least for the first problem, I have granted @gsisinna and @Uboldi80 with push access to the https://github.com/ami-iit/robots-configuration, where we store the leading edge of the configuration files we use on the iCubGenova09 robot.

Agreed 👍🏻 

>For the second problem, my suggestion would be for @Uboldi80 and @gsisinna to add directly in the issue the link to the commit. In this way, it should be easier to recover from similar situations.

Good suggestion 👍🏻 

>For the third problem, I am not sure how to face it in some automatic way.

Difficult point here. Don't think it's easy to get around this problem automatically.
Quick question: when we install the conf files in .local using `yarp-config robots`, does the tool warn you about the presence of existing files that will get overwritten?

S-Dafarra, 2022-02-01:
> Quick question: when we install the conf files in .local using `yarp-config robots`, does the tool warn you about the presence of existing files that will get overwritten?

Actually, I don't know how these tools work. In order to exploit version control, we don't use the ``.local`` folder at all.