robotology/mesh-support #1293: R1SN003 left shouder and left hand issues

elandini84, 2022-02-07:
### Device name 🤖

R1SN003

### Request/Failure description

R1 left sholder seems to have loosened up and the left hand does not work properly.

### Detailed context

As can be seen in the following video the left shoulder of R1SN003 seems to have loosened up
[image]

For the other issue, the left hand of the robot does not starts properly (it starts in `idle` as can be seen from the following video). If, from `yarpmotorgui` the `Run` button is pressed for both the left hand joints, no error is thrown and the encoders seem to work, but it is not possible to move any of them.

https://user-images.githubusercontent.com/62991576/152811616-a0190865-61b2-4721-9800-a3ae0f140f3c.mp4

From 0:53 to 1:16 I tried to gently move the hand joints in order to se if the encoders would have followed and that seems so. From 1:17 onwards, as can be clearly seen, moving the slider does not affect the actual position of the joint.

### Additional context

Regarding the left hand issue, even though I wasn't able to find any useful information in the `yarprobotinterface` log, I'm going to post it here in case you wanted to take a look at it.

[R1SN003_yarprobotinterface_log.txt](https://github.com/robotology/icub-tech-support/files/8016253/R1SN003_yarprobotinterface_log.txt)

emilianob80, 2022-02-09:
The problem with the hand could be caused by this screw which was inside the hand on the MC2 electronic board. It may have caused component U4 to burn out.
It is difficult to tell if it was the EMC test that caused the damage because the right hand was not damaged

[image]
[image]

I am replacing the EB9 electronic board

fbiggi, 2022-02-09:
inizio fase di smontaggio ore 06:00 

[image]

La spalla sx presenta le viti TB 4X16 allentate, procedo a serrare con Frenafiletti

[image]

emilianob80, 2022-02-09:
The card has been replaced and programmed.

[image]

emilianob80, 2022-02-09:
@elandini84 The robot has been repaired. It is necessary to test it before closing the last covers

emilianob80, 2022-02-09:
@elandini84 please can you please correct the title and add right hand and neck

emilianob80, 2022-02-09:
The MC4_PLUS electronic board in package EB10 appeared to have burnt component U4. It has been replaced and scheduled. Unloading in the warehouse carried out

[image]

[image]

emilianob80, 2022-02-09:
After the functional test, the right thumb cord was found to be broken. @fbiggi  is providing for the replacement

emilianob80, 2022-02-10:
@elandini84 robot repaired and working. I proceed with the closure of the issue

closed 2022-02-10 by emilianob80