robotology/mesh-support #1300: iCubErzelli02: right thumb oppose cable replacement

valegagge, 2022-02-14:
### Device name 🤖

iCubErzelli02

### Request/Failure description

The thumb oppose doesn't work properly.

### Detailed context

@Uboldi80 and @gsisinna already saw the problem.

Uboldi80, 2022-02-16:
the tendon was been replaced.
now should be ok, but need to verify the calibration.

valegagge, 2022-02-25:
Yesterday I checked the right arm and I noticed that the thumb doesn't work well as the following video shows.

https://user-images.githubusercontent.com/4233231/155681056-9f3afe5b-3444-41bb-b7e8-a5e667a20aef.mp4

Uboldi80, 2022-03-04:
Also in this case the instability problem was due to an error in the coupling matrix between joints 8 and 9.
The error has been corrected and the files pushed.
https://github.com/robotology/robots-configuration/blob/master/iCubErzelli02/hardware/mechanicals/right_arm-eb28-j8_11-mec.xml.

I imagine that even in this case the problem had been solved locally a long time before and when the laptop and head were redone, the files present in robots-configuration up strem were not suitable for the correct functioning of joints 8 and 9.

closed 2022-03-04 by Uboldi80