robotology/mesh-support #1301: Neck and left thumb tendons broken due to falling

Kattyahs, 2022-02-15:
### Device name 🤖

iCubValparaiso01

### Request/Failure description

Hello, 
While testing robot's movement running demoYoga the piece of metal coupling the pedestal with the robot collapsed and the iCub fell on the floor. Checking the robot, we detected three tensors damaged, two in the neck and another one in the left hand.

### Detailed context

The piece of metal coupling the pedestal with the robot collapsed as you can see in the photo
[image]
We detected two tensors broken in the neck
[image]
And another one in the left thumb 
[image]
After this we haven't turned it on and we only moved the iCub to put it back on the table.
It would be great if you could give us a video call during this week or next one.
Best and thanks for any help.

gsisinna, 2022-02-21:
Good morning @Kattyahs,
thank you for writing in our support and we are sorry for what happened with the robot.

As this is the first time we have encountered a problem with such a component, we would like to better understand the dynamics of the incident and proceed according to your availability to repair the components in question. It is our priority to repair the robot or help you with interventions to restore the system to optimal conditions.

If you are available, we could schedule a call to discuss it together before Friday and study the next steps.

Thank you!

cc @Uboldi @Fabrizio69

mjescobar, 2022-02-21:
Hi,

The video call sounds great! We're available this Thursday during the Chilean morning (Italy afternoon).

Thanks!
María-José

gsisinna, 2022-02-22:
Hi @mjescobar,
for us is fine and I proceed with the creation of the meeting to schedule the call.

Thanks!

cc @Uboldi80

Kattyahs, 2022-02-25:
Hi,
As we talked  yesterday, the pieces we need are the two head cables (head cable "A" and head cable "B") and the opposite thumb cable.
We also noticed that a rubber cover and a plastic protection were broken (See photo). We would like to know if you can send replacement of those parts as well.
[image]
[image]

Thanks!

Kattyahs, 2022-02-28:
In summary the pieces we need are:

- Head Cable "A"
- Head Cable "B"
- Opposite Thumb Cable
- The piece of metal coupling the pedestal with the robot
- A rubber cover
- Plastic protection for left wrist
Thanks!

Uboldi80, 2022-02-28:
Good morning @Kattyahs 
I take note of the spare parts needed to repair the setup and communicate them to the person responsible for shipping.
At the same time I kindly ask you to send us the piece of metal of the pedestal in order to prepare a minimum of internal analysis in order to prevent it from happening again in the future

Kattyahs, 2022-03-02:
Sure, can I ask for the shipping information here?
We need: 

- Name and Surname
- Delivery address
- Postal Code
- Telephone
- Email
- Piece cost (USD)

Uboldi80, 2022-05-18:
After the skin test we noticed a prblem with some wire of pinkie.
we need to replace the wiring to fix the skin of left hand (L3 fingertip)

gsisinna, 2022-05-19:
Today I checked the software of the system, including the laptop configuration. In order:

- The netplan configuration on the icub-head has been restored
- The network interfaces are now configured with the new laptop
- SSH connection checked
- Calibration files have been pushed on robots-configuration repository
- We tested several demos with the new laptop and YARP functionalities

cc @Uboldi80

Uboldi80, 2022-05-23:
I have tested the robot again (all joints) and it seems to work fine.

closed 2022-05-23 by Uboldi80