robotology/mesh-support #1305: Right index and right middle fingertip not working

S-Dafarra, 2022-02-21:
### Device name 🤖

iCubGenova09

### Request/Failure description

The right index and right middle fingertip skin is not working.

### Detailed context

The right index skin is blinking, while the right middle finger seems not working at all.

gsisinna, 2022-02-22:
I replaced an R0 fingertip, soldered two cables, and now they work. Try them out and give feedback please.

I have noticed that during the calibration procedure, based on the starting position, the thumb may conflict with the leg cover. This should be avoided for the safeguard of the fingertips which are very delicate.

cc @Uboldi80

S-Dafarra, 2022-02-24:
Today we tried them. The middle is working fine, but the index is very sensitive. The slightest touch seems enough to make it saturate. Then, it takes a while for it to "discharge"

Uboldi80, 2022-03-09:
i close this one in favour of: 
 https://github.com/robotology/icub-tech-support/issues/1318

closed 2022-03-09 by Uboldi80