robotology/mesh-support #1306: IMU sensors in iCub head

GiuliaScorzaAzzara, 2022-02-21:
### Device name 🤖

iCubGenova03

### Request/Failure description

The MTX inertial box (http://wiki.icub.org/wiki/Inertial_sensing) in iCub head is not working. 

### Detailed context

We opened the head to verify the wire connections and everything seemed to be connected in the proper way, but when starting the robot the inertial sensors in the head did not respond.

Uboldi80, 2022-02-22:
Hy @GiuliaScorzaAzzara 
Tomorrow I will be in Erzelli for an intervention on another robot.
I should still have time to take a look at yours.

Uboldi80, 2022-02-24:
Hy @GiuliaScorzaAzzara, I have the spare part.
I think next Wednesday 2/03/22 i can try to change it to fix the problem.

gsisinna, 2022-02-25:
Today we replaced the MTx intertial box and we fixed also a wrong pinout on the PC104:
[image]

After this intervention it is possible to read the values with yarp from the /icub/interial port:
[image]

Here some test plots for the Euler angles:
[image]

As requested by team members we also fixed a skin patch:
[image]

I would like to thank the team of the laboratory for the collaboration that was essential to identify the problem immediately, and lead to its resolution.

Closing this issue.

cc @Uboldi80 @Fabrizio69 @GiuliaScorzaAzzara

closed 2022-02-25 by gsisinna