robotology/mesh-support #1307: Right thumb oppose doesn't work properly

Uboldi80, 2022-02-22:
The right thumb oppose it's breaking.
it must be replaced.

Uboldi80, 2022-02-23:
today together with giacomo we have tried to solve the problem of the proximal thumb which continues to tear for no apparent reason.
There is probably an error somewhere a tendon wiring error somewhere for a long time.
I probably guess that the oppose thumb might have a wrapping error.
we should replace the tendons of both joints (oppose and proximal) to solve this problem.

gsisinna, 2022-02-25:
Today, together with @Uboldi80 and @GiacomoVassallo13, we continued with the intervention on iCubErzelli01 and we found the cause of the problem at the thumb (oppose and proximal). 

After spending many hours of our technicians on this robot, and doing the thumb wiring at least 5 times, we discovered that in the past had been modified the configuration file contained in `/hardware/mechanicals` regarding the coupling matrix of joints 8 and 9 (J2M matrix), and this change was neither justified nor documented in the history of upstream commits, but probably left locally on the head of the robot and traced only thanks to my backup copy after rebuilding the head software.

As reference, in the official robots-configuration repo ([link](https://github.com/robotology/robots-configuration/blob/master/iCubErzelli01/hardware/mechanicals/right_arm-eb28-j8_11-mec.xml)) we have in `/iCubErzelli01/hardware/mechanicals/right_arm-eb28-j8_11-mec.xml`:

<group name="COUPLINGS"> 
                            
        <param name ="matrixJ2M"> 
            1.000   0.000   0.000   0.000
            1.000  +1.000   0.000   0.000
            0.000   0.000   1.000   0.000
            0.000   0.000   0.000   1.000   
        </param>

         <param name ="matrixM2J"> 
            1.000   0.000   0.000   0.000
           -1.000  +1.000   0.000   0.000
            0.000   0.000   1.000   0.000
            0.000   0.000   0.000   1.000   
        </param>
       
         <param name ="matrixE2J">  
            1.000   0.000   0.000   0.000   0.000   0.000
            0.000   1.000   0.000   0.000   0.000   0.000
            0.000   0.000   1.000   0.000   0.000   0.000
            0.000   0.000   0.000   1.000   0.000   0.000   
        </param>
  
    </group>`

The correct to configuration, in order to have a functioning thumb with **this** robot is:

<group name="COUPLINGS"> 
                            
        <param name ="matrixJ2M"> 
            1.000   0.000   0.000   0.000
           -1.000  +1.000   0.000   0.000
            0.000   0.000   1.000   0.000
            0.000   0.000   0.000   1.000   
        </param>

         <param name ="matrixM2J"> 
            1.000   0.000   0.000   0.000
           +1.000  +1.000   0.000   0.000
            0.000   0.000   1.000   0.000
            0.000   0.000   0.000   1.000   
        </param>
       
         <param name ="matrixE2J">  
            1.000   0.000   0.000   0.000   0.000   0.000
            0.000   1.000   0.000   0.000   0.000   0.000
            0.000   0.000   1.000   0.000   0.000   0.000
            0.000   0.000   0.000   1.000   0.000   0.000   
        </param>
  
    </group>

Not having a commit history, we don't know why it was changed in the past, but we think there was a change in the motor hardware that needed an adjustment to make it all fit together.

Also, looking through the issues, I bring up the following discussion of the above matrix for completeness: https://github.com/robotology/icub-tech-support/issues/1019#issuecomment-730148256

cc @maggia80 @Fabrizio69 @ddetommaso @pattacini @ale-git

gsisinna, 2022-02-25:
For what concern the robot, now the joints calibrate and we tested the setup few times. We will discuss the new configuration before making a pull request.

https://user-images.githubusercontent.com/36999962/155744736-d732a7ee-02b5-4261-b35a-01f90b4968d4.mp4

Clonsing this issue for the thumb, thanks!

cc @ddetommaso @Uboldi80 @GiacomoVassallo13 @maggia80

closed 2022-02-25 by gsisinna

pattacini, 2022-02-25:
>Not having a commit history, we don't know why it was changed in the past, but we think there was a change in the motor hardware that needed an adjustment to make it all fit together.

Most presumably, the signs do account for the way the tendons are winded around the pulleys...
They should be always winded in the same way, but changing the sign in the conf files I know was a quick workaround in case a wrong winding was applied.

Another useful resource explaining those coupling matrices is https://github.com/robotology/robots-configuration/issues/39#issuecomment-324597087.