robotology/mesh-support #1308: The robot's 03 joint cannot be read

tKsome, 2022-02-23:
### Device name 🤖

iCubShenzhenS/N0.39

### Request/Failure description

After repairing the thumb of the robot, we found that the 03 joint motor of the robot was wrong.And there's no change in the encoder.After inspection, it was found that the two power lines of joint 03 were disconnected, and errors were still reported after re-welding.Looking forward to your reply.

### Detailed context

The damaged part is in the robot's right arm

### Additional context

[image]
[image]

Uboldi80, 2022-02-23:
Good morning
the photo you sent related to the current situation?
that is, was the black cable from the hall sensor that should be soldered in the GND position re-soldered correctly?
since from the image on the motorgui that you sent, it seems that there is always a problem in the same point which is probably always due to a wire that has detached from the board with the hall sensor.
generally this can happen when the plastic shell near the shoulder is removed or reassembled which tends to be very wire and the risk is to tear some of it.
for this reason, when the plastic ones are removed and put back, it is necessary to be careful and slightly enlarge them during these operations.
check the hall sensor part of this joint again and try with tweezers to lightly pull the wires to make sure they are still soldered.
if there is glue to protect the welds, remove it carefully. The glue sometimes hides a wire that is actually no longer soldered to the pad.
let me know.

Uboldi80, 2022-02-23:
also check the patch connector upstream of the hall sensor.
also in this case, I recommend to remove the glue carefully and check that the cables are still effectively crimped to the terminals by pulling them lightly with tweezers.

Uboldi80, 2022-02-23:
verificare questo connettore oltre al sensore di hall

[image]

SSSSamZhu, 2022-02-23:
Hi
I compared the two pictures, and I found that one of the encoder chip in the faulty picture seemed to have lost a resistor or capacitor. Could you please tell me the parameters of the relevant components?

Uboldi80, 2022-02-23:
please tell me a photo of the board.

SSSSamZhu, 2022-02-23:
[image]

Uboldi80, 2022-02-23:
ok, yes the capacity is 10uF

[image]

SSSSamZhu, 2022-02-23:
Okay！ I'll see if I can figure it out.Thanks.

Uboldi80, 2022-02-23:
ok, good @tKsome 
after then you will be replaced the capacity, let me know.
THKS!

tKsome, 2022-02-24:
The issue has been resolved and I will close the issue.Thanks

closed 2022-02-24 by tKsome