robotology/mesh-support #1312: Ripped wires in iCub right index finger

GiuliaScorzaAzzara, 2022-02-28:
### Device name 🤖

iCubGenova03

### Request/Failure description

The right index finger of iCub's hand has two ripped wires, which stick out of their proper position.
It may be enough to re-solder the wires. If not, it will be necessary to replace them.

### Detailed context

There are no mechanical issues due to this damage and the robot calibrates itself well when switched on, but the skin sensors on the fingertip of the right index do not respond to tactile stimuli.

Uboldi80, 2022-03-04:
Hy @GiuliaScorzaAzzara 

I think that next Wednesday I will come to Erzelli to try to solve the problem.

Uboldi80, 2022-03-09:
Today, me and @gsisinna  check and fix a loto fo different problems in the right hand:
fixed the solder in R0 (thumb)
fix and replace the R0 (index)
fix and replace the R1 (middle)
furthermore, we fix a short circuit in the ftc connect and a short circuit in the wiring of right index.
Now the right hand is ok and we tested the skin with @lindalastrico.

closed 2022-03-09 by Uboldi80